cyrils/renogy-bt #1: Query - get close to real-time readings

SharkyWaters, 2022-06-08:
Hey I'm new to Python and am trying to log basic info (voltages and currents) from a Renogy BT-1 module as part of a mechanical engineering project at university. What would be the best way to make a loop to get the highest possible resolution out of this code (e.g. 30second intervals)? Any help and guidance will by greatly appreciated! Thank you!

cyrils, 2022-06-09:
Its is possible with little bit of tweaking. I had a branch lying around with real-time monitoring, let me see if I can find it.

SharkyWaters, 2022-06-09:
Thank you very much for checking! I really like the simplicity of your code and only need the basic output. Nonetheless I have tried getting Olen's solar monitor going but have been haven't had any luck yet.

cyrils, 2022-06-09:
Check this branch https://github.com/cyrils/renogy-bt1/tree/real-time. It needs some polishing, I will do it later.

SharkyWaters, 2022-06-09:
Wow thank you! ill be able to access the panel and charge controller again early next week, so ill give it another shot then and let you know how it goes!

SharkyWaters, 2022-07-13:
Hey! Just wanted to let you know that I finally managed to test the new code for real time monitoring. Works perfectly down to 0.1sec intervals! Thank you so so much for the help!

cyrils, 2022-07-13:
Glad that helped. Hope you are not stressing the device too much with 0.1 sec polling.

SharkyWaters, 2022-07-13:
Yeah, I was worried the Pi might not be able to handle it, but I've run the logging for a few minutes and it held up perfectly! Only have an issue every now and then where it fails to establish the bluetooth connection, but turning the bluetooth on the Pi off and on and/or restarting Thonny seems to do the trick :)

closed 2023-07-24 by cyrils