cyrils/renogy-bt #4: negative sensor temperature

Byter09, 2022-12-15:
Hello!

First of, I would like to thank you for this repository. I use it to send off data updates to InfluxDB and Home Assistant :)

Now, to the problem. Its getting quite cold, and the temperature at the battery probe went below 0°C.

It's unclear whether this is the Rover 20a having a problem, or how the data is retrieved. Just to make sure I'm opening this ticket. I checked both InfluxDB and Home Assistant and they both report the same value. I then went ahead and checked the webhook I use to receive the data I'm sending with your neat project, and it doesn't seem to change any negative values (I sent mock data to check and it works perfectly), so that leaves me to believe it might be a bug in your code. If I knew more about python I'd have a look but it's not really my forte haha.

The resulting graph from last night looks like this haha
[image]

It went up to 129°C, which looks like an underflow to me.

If you need any more information, I'm happy to help out with whatever you need.

cyrils, 2022-12-15:
Could be problem with hex parsing, I'll look into it later tonight.

cyrils, 2022-12-16:
@Byter09 I've updated the code, can you check it out? According to documentation 7th bit it for sign, which I've been ignoring. Only change is in `Utils.py`.

Byter09, 2022-12-16:
I have updated my docker container and I guess we'll see if it works. But thank you for taking the time. If you want me I can put some ice cubes on it haha :D

closed 2023-07-24 by cyrils