cyrils/renogy-bt #11: Connection failed

deltaprise, 2023-05-07:
Device: Renogy Rover LI 40A
Bluetooth-Modul BT-1

Log from starting the ./example.py

INFO:root:Starting client: BT-TH-7749A78E => f4:60:77:49:a7:8e
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 9
INFO:root:Found matching device BT-TH-7749A78E     => [f4:60:77:49:a7:8e]
INFO:root:[f4:60:77:49:a7:8e] Discovered, alias = BT-TH-7749A78E
INFO:root:[f4:60:77:49:a7:8e] Discovered, alias = BT-TH-7749A78E
INFO:root:[f4:60:77:49:a7:8e] Connected
INFO:root:[f4:60:77:49:a7:8e] Discovered, alias = BT-TH-7749A78E
INFO:root:[f4:60:77:49:a7:8e] Disconnected
ERROR:root:Connection failed: Disconnected

The mac address an the device alias is correct, why doesn't it work?

cyrils, 2023-05-07:
Usually this happens when bluetooth signal is weak. Are you close enough to the device?

My own personal system also uses Rover 40A and BT-1 module and it works without any issue.

deltaprise, 2023-05-07:
It was the weak bluetooth signal, thanks

closed 2023-05-07 by deltaprise