cyrils/renogy-bt #12: Home assistant Renogy on PI 4 error

marcuslib, 2023-07-12:
Hi hope you are well, i have been trying to do this for weeks now, please help

I have installed Home Assistant using the Raspberry Pi imager 32bit version. i have a Pi 4

I can't seem to get it to work, everytime I run python3 ./example.py, 

it says I don't have something installed, when I run python3 -m pip install -r requirements.txt, 

libscrc errors out, then managed to get that installed, then says Module 'dbus' not found is missing, manage to install that, then it say ModuleNotFoundError: No module named 'gi'

cant install any further, then when I reboot, it looses everything and says nothing is installed and I have to start again.

can I check how you are installing Home assistant on the pi, maybe I'm doing that start bit wrong. 

Thanks for any help

cyrils, 2023-07-13:
Hello, 
I haven't tested the code with Home Assistant image, its designed primarily to run on Raspberry Pi OS, and then it sends data to Home Assistant using MQTT protocol.

marcuslib, 2023-07-13:
So if I install it within docker and then send it that way, would that work, worry I'm new to this. --Sent from my Android phone with mail.com Mail. Please excuse my brevity.

cyrils, 2023-07-13:
Yeah that might work, it makes a clean install on new raspberry image. Not sure how the IP address looks like from docker though. But ideally a dedicated hardware is better for HA.

marcuslib, 2023-07-17:
i have now got it working in docker, I keep getting ERROR:root:Connection failed: le-connection-abort-by-local when running python3 ./example.py, some times it works, due to know what this might be, is this because its already connected. it not connected to anything else.

marcuslib, 2023-07-17:
(venv) marcuslibby@raspberry:/opt/path/to/venv/bin/renogy$ python3 ./example.py
INFO:root:Starting client: BT-TH-161CA41D => ac:4d:16:1c:a4:1d
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 8
INFO:root:Found matching device BT-TH-161CA41D     => [ac:4d:16:1c:a4:1d]
ERROR:root:Connection failed: le-connection-abort-by-local

cyrils, 2023-07-17:
Are you close enough to the device?

marcuslib, 2023-07-17:
Yes, within 10cm, they are next to each other. It did work for a second and reported into home assistant, then just stopped. --Sent from my Android phone with mail.com Mail. Please excuse my brevity.

cyrils, 2023-07-17:
Is the `enable_polling` flag set to true in config.ini?

marcuslib, 2023-07-17:
it seems to work after about 30 tries of running "python3 ./example.py" within terminal

marcuslib, 2023-07-19:
it seems to connect now and run, how do I get it to auto-run all the time without having to manually run the example file. thanks again

cyrils, 2023-07-20:
If polling is enabled it should keep running. If you want to auto-start it on restarts add it to `/etc/rc.local`. Or you can disable polling and schedule a periodic cron job using crontab.

marcuslib, 2023-07-21:
im not very good with Python, 

/etc/rc.local does not really work, as i want it running all the time, and if it stops, I want it to restart, I tried cron job, but cant get that to run; the only way really is to run the code below in terminal. 

source /opt/path/to/venv/bin/activate && cd /opt/path/to/venv/bin/renogy && python3 ./example.py >> /opt/path/to/venv/bin/renogy/log_file.log 2>&1

cyrils, 2023-07-21:
Run `crontab -e` and insert the following command:
```shell
*/5 * * * * /opt/path/to/venv/bin/renogy/example.py 
```
This will run every 5 minutes (turn off polling in config.ini)

marcuslib, 2023-07-23:
That does not seem to work, it only works when I run the full script. I'm running unbunto, docker and portainor, I can only get it to run on a virtual venv. --Sent from my Android phone with mail.com Mail. Please excuse my brevity.

cyrils, 2023-07-23:
Yes, you can replace it with the full command. But you get the idea, */5 means it runs every five minutes.

marcuslib, 2023-07-24:
Finally got it working, had to add it as */5 * * * * /opt/path/to/venv/bin/python3 /opt/path/to/venv/bin/renogy/example.py

marcuslib, 2023-07-25:
Did not work, as it kept trying to run again and cancel the previous one; in the end, I set up a service that runs the script at the start and if it stops, restarts it. works a treat. 

Copied this https://gist.github.com/emxsys/a507f3cad928e66f6410e7ac28e2990f

closed 2023-08-23 by cyrils