cyrils/renogy-bt #13: Device not found BT-1 Bluetooth Module (New Version)

Nkaati, 2023-07-18:
I can see the device in Linux when running bluetoothctl
scan on.
[NEW] Device 60:98:66:EF:9D:A6 60-98-66-EF-9D-A6

But when I run the example.py
python example.py
INFO:root:Starting client: BT-TH-66EF9DA6 => 60:98:66:EF:9D:A6
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 2
INFO:root:Devices found: 7
INFO:root:Devices found: 10
INFO:root:Devices found: 10
INFO:root:Devices found: 12
ERROR:root:Device not found: BT-TH-66EF9DA6 => 60:98:66:EF:9D:A6, please check the details provided.

Wondered if there is something missing with this newer BT1 module.  I have it connected to a Wander 30A Controller

cyrils, 2023-07-19:
Usually this happens when:
1. Device is already connected/paired to some other client. Reconnecting the bt-1 or exiting any other bluetooth app fixes this.
2. The device is not close enough to the client, bluetooth signal is weak.

I don't see really any other reason since it shows up in your OS

pwteneyck, 2023-07-22:
I was having a similar issue and realized that the default device alias, `renogy-bt1`, actually had a bunch of whitespace after it: `renogy-bt1      `. I tried various different "fixes", like adding `strip()` calls to the code in some places or removing them from others, but in the end the only thing that worked for me was to use `bluetoothctl` to give the device a different alias without the trailing whitespace:

```
$ sudo bluetoothctl
[bluetooth]# scan on
[bluetooth]# connect <BT1_MAC_ADDRESS>
[bluetooth]# set-alias renogy-bt1
[bluetooth]# disconnect renogy-bt1
[bluetooth]# quit
```

Nkaati, 2023-07-23:
Ah, thank you pwteneyck.  You got me thinking about what the actual alias was. 
From bluetoothctl ->
[NEW] Device 60:98:66:EF:9D:A6 60-98-66-EF-9D-A6

60-98-66-EF-9D-A6  was the alias and that worked to connect via the python script!

cyrils, 2023-07-23:
Good to hear it worked out for both of you. But it actually doesn't make sense as alias is not mandatory. It should connect just using `mac_addr`. If any one of you can share your `config.ini` may be I can try to find the root cause.

pwteneyck, 2023-07-23:
I suspect the problems are deeper than the code in this repo. The "obvious" cause is [here](https://github.com/cyrils/renogy-bt1/blob/main/renogybt/BLE.py#L29):

```
if (dev.mac_address == self.mac_address or dev.alias() == self.device_alias)
  ...
```

Where I'm fairly certain that Python's `configparse` automatically strips leading and trailing whitespace from config values. That means that it's impossible to use `configparse` to supply an alias that matches `BT-TH-9B236ACB    `. With that in mind, I tried hardcoding the correct alias (whitespace included) into `example.py` and started running into different issues. I don't remember what they were off the top of my head.

### Repro steps (maybe?)

1. Set target device alias to something with trailing whitespace

```
$ sudo bluetoothctl
[bluetooth]# scan on
[bluetooth]# connect <BT1_MAC_ADDRESS>
[bluetooth]# set-alias 'one_trailing_space '
[bluetooth]# disconnect <BT1_MAC_ADDRESS>
[bluetooth]# exit
```

2. Set the "proper" MAC addr and alias in `config.ini`:

```
[device]
adapter = hci0
mac_addr = <BT1_MAC_ADDRESS>

# make sure to include the trailing space here ;)
alias = one_trailing_space 
...
```

3. Try it out

```
$ python example.py
```

cyrils, 2023-07-23:
@pwteneyck Yes you are right, `configparser` strips out trailing spaces, but as you can see here:

```python
if (dev.mac_address == self.mac_address or dev.alias() == self.device_alias)
```

which means even if alias doesn't match, it should match the `mac_address`. Any chance you have the `mac_addr` in your config in lower case?

Nkaati, 2023-07-23:
Nope, it was all uppercase.  I attached the file in the comment above.  I'll try a few more variations to see what breaks it from connecting versus connects okay.

cyrils, 2023-07-23:
@Nkaati Okay, at least in you case the mac address detected is in lowercase which explains why the match failed. Probably I should add case insensitive check.

Nkaati, 2023-07-23:
@cyrils Yep, if I leave the alias blank and just use a lowercase MAC address it works.  So just case sensitivity.  Thank you.

cyrils, 2023-07-24:
I have a fix for this in latest master.

closed 2023-08-23 by cyrils