cyrils/renogy-bt #15: Bluetooth Lifepo4 batteries

sven253, 2023-07-30:
Hello,
will you support renogy bluetooth batteries? https://github.com/chadj/renogy-smart-battery this is working but is webbased. i have no idea how your script or the other works  but i can test it if you want to support batteries, too.

cyrils, 2023-07-30:
Sure, it might take several iterations, but I'm happy to help. I'll let you know when there is a branch ready to test.

cyrils, 2023-07-30:
@sven253 Please checkout the branch [battery](https://github.com/cyrils/renogy-bt1/tree/battery), modify `config.ini`, run `example_battery.py` and let me know the output log. If it doesn't work try changing the `DEVICE_ID` in `BatteryClient.py` to 247 (or whatever your app shows).

Please note this is super early version and it may have many bugs.

sven253, 2023-07-31:
thanks for your fast response! I used DEVICE_ID 48
`INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 17
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_cell_voltage_payload 5000 => [48, 3, 19, 136, 0, 11, 132, 130]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/zero/renogy-bt1-battery/./example_battery.py", line 20, in <module>
    BatteryClient(config, on_data_received).connect()
  File "/home/zero/renogy-bt1-battery/renogybt/BatteryClient.py", line 38, in connect
    self.manager.run()
  File "/home/zero/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 91, in run
    self._main_loop.run()
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 497, in run
    super(MainLoop, self).run()
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 537, in _signal_func
    match.maybe_handle_message(message)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/zero/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=bytes(value))
  File "/home/zero/renogy-bt1-battery/renogybt/BLE.py", line 86, in characteristic_value_updated
    self.data_callback(value)
  File "/home/zero/renogy-bt1-battery/renogybt/BatteryClient.py", line 58, in __on_data_received
    logging.info("on_data_received: response for read operation", value)
Message: 'on_data_received: response for read operation'
Arguments: (b'0\x03\x16\x00\x04\x00!\x00!\x00!\x00!\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00,y',)
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ_VOLTS', 'cell_count': 48, 'cellvoltage_1': 22, 'cellvoltage_2': 4, 'cellvoltage_3': 33, 'cellvoltage_4': 33, 'cellvoltage_5': 33, 'cellvoltage_6': 33, 'cellvoltage_7': 0, 'cellvoltage_8': 0, 'cellvoltage_9': 0, 'cellvoltage_10': 0, 'cellvoltage_11': 0, 'cellvoltage_12': 0, 'cellvoltage_13': 121, 'cellvoltage_14': 0, 'cellvoltage_15': 0, 'cellvoltage_16': 0, 'cellvoltage_17': 0, 'cellvoltage_18': 0, 'cellvoltage_19': 0, 'cellvoltage_20': 0, 'cellvoltage_21': 0, 'cellvoltage_22': 0, 'cellvoltage_23': 0, 'cellvoltage_24': 0, 'cellvoltage_25': 0, 'cellvoltage_26': 0, 'cellvoltage_27': 0, 'cellvoltage_28': 0, 'cellvoltage_29': 0, 'cellvoltage_30': 0, 'cellvoltage_31': 0, 'cellvoltage_32': 0, 'cellvoltage_33': 0, 'cellvoltage_34': 0, 'cellvoltage_35': 0, 'cellvoltage_36': 0, 'cellvoltage_37': 0, 'cellvoltage_38': 0, 'cellvoltage_39': 0, 'cellvoltage_40': 0, 'cellvoltage_41': 0, 'cellvoltage_42': 0, 'cellvoltage_43': 0, 'cellvoltage_44': 0, 'cellvoltage_45': 0, 'cellvoltage_46': 0, 'cellvoltage_47': 0}
INFO:root:Exit: Disconnecting device: BT-TH-161E1B29 [AC:4D:16:1E:1B:29]`

if i use 247 i dont get any output after connecting. where can i see the DEVICE_ID of my device?

cyrils, 2023-08-01:
I've updated the branch, lets try this again.

sven253, 2023-08-01:
In the Renogy DC Home App?
I got this output:

`INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 20
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_cell_voltage_payload 5000 => [48, 3, 19, 136, 0, 46, 69, 89]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.9/logging/__init__.py", line 1079, in emit
    msg = self.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 923, in format
    return fmt.format(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 659, in format
    record.message = record.getMessage()
  File "/usr/lib/python3.9/logging/__init__.py", line 363, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
Call stack:
  File "/home/zero/renogy-bt1-battery/./example_battery.py", line 20, in <module            >
    BatteryClient(config, on_data_received).connect()
  File "/home/zero/renogy-bt1-battery/renogybt/BatteryClient.py", line 38, in co            nnect
    self.manager.run()
  File "/home/zero/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line             91, in run
    self._main_loop.run()
  File "/usr/lib/python3/dist-packages/gi/overrides/GLib.py", line 497, in run
    super(MainLoop, self).run()
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 537, in _signal            _func
    match.maybe_handle_message(message)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_h            andle_message
    self._handler(*args, **kwargs)
  File "/home/zero/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line             539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=            bytes(value))
  File "/home/zero/renogy-bt1-battery/renogybt/BLE.py", line 86, in characterist            ic_value_updated
    self.data_callback(value)
  File "/home/zero/renogy-bt1-battery/renogybt/BatteryClient.py", line 58, in __            on_data_received
    logging.info("on_data_received: response for read operation", value.hex())
Message: 'on_data_received: response for read operation'
Arguments: ('30035c0004002100210021002100000000000000000000000000000000000000000            0000000000200d200d20000000000000000000000000000000000000000000000000000000000000            00000000000000000000000000000000085000161e89c61',)
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ', 'cell_count': 4, 'cell_voltage            _0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2':             3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'cell_temperature_0':             0.0, 'cell_temperature_1': 0.0, 'cell_temperature_2': 0.0, 'cell_temperature_3'            : 0.0, 'current': 0.0, 'voltage': 0.0, 'remaining_charge': 8716.289, 'capacity':             1642634.337}
INFO:root:Exit: Disconnecting device: BT-TH-161E1B29 [AC:4D:16:1E:1B:29]
`

cyrils, 2023-08-01:
I haven't tried the Home DC app as I don't have a battery. The renogy bt app shows device Id in the 'device info' tab.

Looks like its reading the cell voltages correctly. I've pushed in some fixes for remaining fields, let me know if its working.

sven253, 2023-08-01:
I tried the Renogy App but it doesnt support batteries.  I dont get any battery information. In the device info tab theres is only my battery name.
I tried your latest push it works! Thank you!
`
INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 17
INFO:root:Found matching device BT-TH-161E1B29 => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_cell_voltage_payload 5000 => [48, 3, 19, 136, 0, 48, 197, 81]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ', 'cell_count': 4, 'sensor_count': 2, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'cell_temperature_0': 21.0, 'cell_temperature_1': 21.0, 'current': 0.0, 'voltage': 13.3, 'remaining_charge': 90.60000000000001, 'capacity': 100.0}
INFO:root:Exit: Disconnecting device: BT-TH-161E1B29 [AC:4D:16:1E:1B:29]
`

cyrils, 2023-08-01:
Great, I will do a refactor later sometime to push these changes to master , you can continue to use the branch in the mean time.

Can you also do a favor let me know the exact battery model you have? I could add that to the list of supported devices. Some other battery versions seems to have a different mapping of fields (mentioned here [Grmume/renogy-smart-battery](https://github.com/Grmume/renogy-smart-battery)) not sure how is it different from your model.

sven253, 2023-08-01:
I have this one:
RBT100LFP12-BT-DE 
https://de.renogy.com/12-v-100-ah-lithium-lifepo4-akku-mit-bluetooth/
Should be the same as this one:
RBT100LFP12-BT-US
https://www.renogy.com/12v-100ah-lithium-iron-phosphate-battery-w-bluetooth/

sven253, 2023-08-02:
i found a little bug:
current: 653.96
It should be 1,5 A
Can you try to fix it?

cyrils, 2023-08-03:
Can you replace `BatteryClient.py#59` with the following and let me know the output?

```python
logging.info(f"on_data_received: response for read operation {value.hex()}")
```

sven253, 2023-08-03:
current = 0 A
`
INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 18
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_read_request 5000 => [48, 3, 19, 136, 0, 48, 197, 81]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation 30036000040021002100210021000000000000000000000000000000000000000000000000000200d200d20000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000850001449c000186a09606
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ', 'cell_count': 4, 'sensor_count': 2, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'temperature_0': 21.0, 'temperature_1': 21.0, 'current': 0.0, 'voltage': 13.3, 'remaining_charge': 83.10000000000001, 'capacity': 100.0}
INFO:root:mqtt logging
^CERROR:root:Exception occured: KeyboardInterrupt
INFO:root:Exit: Disconnecting device: BT-TH-161E1B29 [AC:4D:16:1E:1B:29]
`

approx current 1,5 A
`
INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 22
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_read_request 5000 => [48, 3, 19, 136, 0, 48, 197, 81]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation 30036000040021002100210021000000000000000000000000000000000000000000000000000200d200d20000000000000000000000000000000000000000000000000000000000000000000000000000000000000000ff7400850001449c000186a07ee2
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ', 'cell_count': 4, 'sensor_count': 2, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'temperature_0': 21.0, 'temperature_1': 21.0, 'current': 653.96, 'voltage': 13.3, 'remaining_charge': 83.10000000000001, 'capacity': 100.0}
INFO:root:mqtt logging
`

cyrils, 2023-08-03:
Should be fixed now in latest commit.

sven253, 2023-08-03:
Thank you for your fast response! I tried you latest commit but got the same value:
`
INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 18
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 48, 197, 81]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:BT-TH-161E1B29 => {'function': 'READ', 'cell_count': 4, 'sensor_count': 2, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'temperature_0': 21.0, 'temperature_1': 21.0, 'current': 653.96, 'voltage': 13.200000000000001, 'remaining_charge': 81.4, 'capacity': 100.0}
INFO:root:mqtt logging
`

cyrils, 2023-08-03:
Lolz.. Pushed everything but the the fix. Can you pull again?

sven253, 2023-08-03:
With your latest commit i doesnt get any data from the batty. This is my output:
INFO:root:Starting client: BT-TH-161E1B29 => AC:4D:16:1E:1B:29
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 16
INFO:root:Found matching device BT-TH-161E1B29     => [ac:4d:16:1e:1b:29]
INFO:root:[AC:4D:16:1E:1B:29] Connected
INFO:root:[AC:4D:16:1E:1B:29] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded

cyrils, 2023-08-03:
Must be a temporary issue. Last commit is just a single line change https://github.com/cyrils/renogy-bt1/commit/5f05364eeaa505367bbde95152277b966c0f0c24

sven253, 2023-08-04:
Sorry my fault. i used the wrong python script. Everything works great! Thank you!!!

sven253, 2023-08-06:
its a little bit off topic but i changed your code a litte bit to fit my needs. I want to search for bluetooth devices till i find my battery because my battery is portable and not allways in the bluetooth range of my pi. i created a service to call your script and start it when automaticaly with pi. my python skills are pritty low. Can you take a look at it? its working but i am wondering if this is a good way to do it.

This is the part i changed:
BLE.py
` 
NONSTOP_DISCOVERY_TIMEOUT = 5 * 60 # time to wait till new search
   def discover(self):
        discovering = True; wait = DISCOVERY_TIMEOUT; self.device_found = False; mac_address = self.mac_address.upper();

        self.update_devices()
        logging.info("Starting discovery...")
        self.start_discovery()

        while discovering:
            time.sleep(1)
            logging.info("Devices found: %s", len(self.devices()))
            for dev in self.devices():
                if dev.mac_address != None and (dev.mac_address.upper() == mac_address or dev.alias() == self.device_alias) and discovering:
                    logging.info("Found matching device %s => [%s]", dev.alias(), dev.mac_address)
                    discovering = False; self.device_found = True
            wait = wait -1
            if (wait <= 0):
                self.stop_discovery()
                logging.error(f"No device found wait {NONSTOP_DISCOVERY_TIMEOUT} seconds")
                time.sleep(NONSTOP_DISCOVERY_TIMEOUT)
                self.discover()
        self.stop_discovery()
`

cyrils, 2023-08-06:
It should work, but might lead to a never ending loop in case battery is not reachable. The method I have is a little different, since I do not want real-time data.

I have turned off polling in config.ini and scheduled a cronjob (`crontab -e`) like this :
```sh
*/5 * * * * /path/to/renogy/example.py 
```

This will run every 5 minutes.

pedrong1025, 2023-08-14:
Hi all, I got a Renogy 100AH Smart Battery (RBT100LFP12S-BT-UK) with the BT-2 module. If 48 is used as Device ID, the output is 37182 cells with all zeros. No output if 247 is used. Just wonder what's the correct Device ID? I can't find such info in Renogy BT app. Any help is greatly appreciated. Many thanks!

cyrils, 2023-08-15:
@pedrong1025 Currently only battery with built-in bluetooth is supported. However if you can share the logs may be I can take a look.

Update `BatteryClient.py#31` with following code and let me know the output:
```py
logging.info(f"on_data_received: response for read operation {value.hex()}")
```

pedrong1025, 2023-08-15:
Hi Cyrils, thanks for your reply! Here is the log. Once again, thank you very much for your help!

INFO:root:Starting client: BT-TH-9B2..... => FC:A8:9B:.......
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 19
INFO:root:Devices found: 21
INFO:root:Found matching device BT-TH-9B....     => [fc:a8:9b:........]
INFO:root:[FC:A8:9B:2...] Connected
INFO:root:[FC:A8:9B:....] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 48, 197, 81]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:Modified - on_data_received: response for read operation 308302913e
DEBUG:root:BT-TH-........... => {'function': None, 'cell_count': 37182, 'sensor_count': 0, 'cell_voltage_0': 0.0, 'cell_voltage_1': 0.0, 'cell_voltage_2': 0.0, 'cell_voltage_3': 0.0, 'cell_voltage_4': 0.0, 'cell_voltage_5': 0.0, 'cell_voltage_6': 0.0, 'cell_voltage_7': 0.0, 'cell_voltage_8': 0.0, 'cell_voltage_9': 0.0, 'cell_voltage_10': 0.0, 'cell_voltage_11': 0.0, 'cell_voltage_12': 0.0, 'cell_voltage_13': 0.0, 'cell_voltage_14': 0.0, 'cell_voltage_15': 0.0, 'cell_voltage_16': 0.0, 'cell_voltage_17': 0.0, 'cell_voltage_18': 0.0, 'cell_voltage_19': 0.0, 'cell_voltage_20': 0.0, 'cell_voltage_21': 0.0, 'cell_voltage_22': 0.0, 'cell_voltage_23': 0.0, 'cell_voltage_24': 0.0, 'cell_voltage_25': 0.0, 'cell_voltage_26': 0.0, 'cell_voltage_27': 0.0, 'cell_voltage_28': 0.0, 'cell_voltage_29': 0.0, 'cell_voltage_30': 0.0, 'cell_voltage_31': 0.0, 'cell_voltage_32': 0.0, 'cell_voltage_33': 0.0, 'cell_voltage_34': 0.0, 'cell_voltage_35': 0.0, 'cell_voltage_36': 0.0, 'cell_voltage_37': 0.0, 'cell_voltage_38': 0.0, 'cell_voltage_39': 0.0, 'cell_voltage_40': 0.0, 'cell_voltage_41': 0.0, 'cell_voltage_42': 0.0, 'cell_voltage_43': 0.0, 'cell_voltage_44': 0.0, 'cell_voltage_45': 0.0, 'cell_voltage_46': 0.0, 'cell_voltage_47': 0.0, 'cell_voltage_48': 0.0, 'cell_voltage_49': 0.0, 'cell_voltage_50': 0.0, 'cell_voltage_51': 0.0, 'cell_voltage_52': 0.0, 'cell_voltage_53': 0.0, 'cell_voltage_54': 0.0

cyrils, 2023-08-16:
Unfortunately that didn't give much information. Can you try changing the following in `BatteryClient#14` as well and let me know the output log:

```py
READ_PARAMS = {
    'FUNCTION': 3,
    'REGISTER': 136,
    'WORDS': 30
}
```

pedrong1025, 2023-08-16:
Hi Cyrils, thanks again for the feedback! I changed the code but unfortunately got the same output. Anything else I can modify to make it more informative?

root@ha-vm01:/opt/renogy-bt-main/renogybt# python3 /opt/renogy-bt-main/battery.py 
INFO:root:Starting client: BT-TH-xxxxxxx => FC:A8:xxxxxxx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 17
INFO:root:Found matching device BT-TH-xxxxxxx     => [fc:a8:xxxxxxx]
INFO:root:[FC:A8:xxxxxxx] Connected
INFO:root:[FC:A8:xxxxxxx] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_request_payload 136 => [48, 3, 0, 136, 0, 30, 65, 201]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:Modified - on_data_received: response for read operation 308302913e
DEBUG:root:BT-TH-xxxxxxx => {'function': None, 'cell_count': 37182, 'sensor_count': 0, 'cell_voltage_0': 0.0, 'cell_voltage_1': 0.0, 'cell_voltage_2': 0.0, 'cell_voltage_3': 0.0, 'cell_voltage_4': 0.0, 'cell_voltage_5': 0.0, 'cell_voltage_6': 0.0, 'cell_voltage_7': 0.0, 'cell_voltage_8': 0.0, 'cell_voltage_9': 0.0, 'cell_voltage_10': 0.0, 'cell_voltage_11': 0.0, 'cell_voltage_12': 0.0, 'cell_voltage_13': 0.0, 'cell_voltage_14': 0.0, 'cell_voltage_15': 0.0, 'cell_voltage_16': 0.0, 'cell_voltage_17': 0.0, 'cell_voltage_18': 0.0, 'cell_voltage_19': 0.0, 'cell_voltage_20': 0.0, 'cell_voltage_21': 0.0, 'cell_voltage_22': 0.0, 'cell_voltage_23': 0.0, 'cell_voltage_24': 0.0, 'cell_voltage_25': 0.0, 'cell_voltage_26': 0.0, 'cell_voltage_27': 0.0, 'cell_voltage_28': 0.0, 'cell_voltage_29': 0.0, 'cell_voltage_30': 0.0, 'cell_voltage_31': 0.0, 'cell_voltage_32': 0.0, 'cell_voltage_33': 0.0, 'cell_voltage_34': 0.0, 'cell_voltage_35': 0.0, 'cell_voltage_36': 0.0,

pedrong1025, 2023-08-17:
Hi Cyrils, after changing the code to, 

READ_PARAMS = {
    'FUNCTION': 3,
    'REGISTER': 5000,
    'WORDS': 17.  ### The original code is 48
}

I got more informative output. Is this you are looking for? 

INFO:root:Modified - on_data_received: response for read operation 30032200040021002100210021000000000000000000000000000000000000000000000000d035
DEBUG:root:BT-TH-xxxxxxx => {'function': 'READ', 'cell_count': 4, 'sensor_count': 53301, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.3000000000000003, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'temperature_0': 0.0, 'temperature_1': 0.0, 'temperature_2': 0.0, 'temperature_3': 0.0, 'temperature_4': 0.0, 'temperature_5': 0.0, 'temperature_6': 0.0, 'temperature_7': 0.0, 'temperature_8': 0.0, 'temperature_9': 0.0, 'temperature_10': 0.0, 'temperature_11': 0.0, 'temperature_12': 0.0, 'temperature_13': 0.0, 'temperature_14': 0.0, 'temperature_15': 0.0, 'temperature_16': 0.0, 'temperature_17': 0.0, 'temperature_18': 0.0, 'temperature_19': 0.0, 'temperature_20': 0.0, 'temperature_21': 0.0, 'temperature_22': 0.0, 'temperature_23': 0.0, 'temperature_24': 0.0, 'temperature_25': 0.0, 'temperature_26': 0.0, 'temperature_27': 0.0, 'temperature_28': 0.0, ' ......

Regarding your questions, 

1. Yes, I got a controller. I can get voltage, temperature ...etc by using your code. 
2. No, I don't use any communication hub, just use BT1 for the controller and BT2 for the battery

Thank you very much for your help!

Pedro

pedrong1025, 2023-08-17:
Hi Cyrils, if I change the words to 24, I get more meaningful output. But the value of current, voltage, remaining charge and capacity are zero.

INFO:root:Starting client: BT-TH-xxxxxx => FC:A8:9Bxxxxx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 17
INFO:root:Found matching device BT-TH-xxxxxx     => [fc:a8:9b:xxxxxx]
INFO:root:[FC:A8:9B:20:6A:1B] Connected
INFO:root:[FC:A8:9B:20:6A:1B] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:reading params
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 24, 197, 79]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:Modified - on_data_received: response for read operation 30033000040024002400240024000000000000000000000000000000000000000000000000000400dc00dc00dc00dcaeec0000403d
DEBUG:root:BT-TH-9B206A1B => {'function': 'READ', 'cell_count': 4, 'sensor_count': 4, 'cell_voltage_0': 3.6, 'cell_voltage_1': 3.6, 'cell_voltage_2': 3.6, 'cell_voltage_3': 3.6, 'temperature_0': 22.0, 'temperature_1': 22.0, 'temperature_2': 22.0, 'temperature_3': 22.0, 'current': 0.0, 'voltage': 0.0, 'remaining_charge': 0.0, 'capacity': 0.0}
INFO:root:mqtt logging
INFO:root:Exit: Disconnecting device: BT-TH-9Bxxxxx [FC:A8:9B:xxxxx]

cyrils, 2023-08-17:
I see, makes sense. Looks like BT-2 doesn't allow long read values. I might have to modify the code to read it in batches.

cyrils, 2023-08-18:
@pedrong1025 Can you try [battery-bt2](https://github.com/cyrils/renogy-bt/tree/battery-bt2) branch and let me know?

pedrong1025, 2023-08-19:
Hi Cyrils, the result looks good to me. Many thanks for your help!

INFO:root:Starting client: BT-TH-xxxxxxx => FC:A8:xxxxxxx
INFO:root:Initing client: BT-TH-xxxxxxx => FC:A8:xxxxxxx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 18
INFO:root:Found matching device BT-TH-xxxxxxx     => [fc:a8:xxxxxxx]
INFO:root:[FC:A8:xxxxxxx] Connected
INFO:root:[FC:A8:xxxxxxx] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 24, 197, 79]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation 30033000040024002400240024000000000000000000000000000000000000000000000000000400d200d200d200d2aeec0000beb4
DEBUG:root:create_request_payload 5042 => [48, 3, 19, 178, 0, 6, 101, 74]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation 30030c0000009100018665000186a0b344
DEBUG:root:create_request_payload 5122 => [48, 3, 20, 2, 0, 8, 228, 29]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation 3003105242543130304c46503132532d47310091e8
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-xxxxxxx => {'function': 'READ', 'cell_count': 4, 'cell_voltage_0': 3.6, 'cell_voltage_1': 3.6, 'cell_voltage_2': 3.6, 'cell_voltage_3': 3.6, 'sensor_count': 4, 'temperature_0': 21.0, 'temperature_1': 21.0, 'temperature_2': 21.0, 'temperature_3': 21.0, 'current': 0.0, 'voltage': 14.5, 'remaining_charge': 99.941, 'capacity': 100.0, 'model': 'RBT100LFP12S-G'}
INFO:root:Exit: Disconnecting device: BT-TH-xxxxxxx [FC:A8:xxxxxxx]

cyrils, 2023-08-19:
I did some refactoring to make it more robust. Can you pull from `battery-bt2` branch again and let me know if tis working? I will port it to master depending on that. (`config.ini` has a new `type` field now)

pedrong1025, 2023-08-21:
Hi Cyrils, sorry for late reply!

I tried to run the new code but it doesn't work. Here is the output.

root@ha-vm01:/tmp/bt2# python3 ./example.py 
INFO:root:Init BatteryClient: BT-TH-xxxxxxx => FC:A8:xxxxxxx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 23
INFO:root:Found matching device BT-TH-xxxxxxx     => [fc:a8:xxxxxxx]
INFO:root:[FC:A8:xxxxxxx] Connected
INFO:root:[FC:A8:xxxxxxx] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 17, 5, 73]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation30032200040024002400240024000000000000000000000000000000000000000000000000c1e4
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/gatt/gatt_linux.py", line 539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=bytes(value))
  File "/tmp/bt2/renogybt/BLE.py", line 86, in characteristic_value_updated
    self.data_callback(value)
  File "/tmp/bt2/renogybt/BaseClient.py", line 62, in on_data_received
    self.data.update(parsed_data)
TypeError: 'NoneType' object is not iterable
INFO:root:on_data_received: response for read operation30032200040024002400240024000000000000000000000000000000000000000000000000c1e4
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/gatt/gatt_linux.py", line 539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=bytes(value))
  File "/tmp/bt2/renogybt/BLE.py", line 86, in characteristic_value_updated
    self.data_callback(value)
  File "/tmp/bt2/renogybt/BaseClient.py", line 62, in on_data_received
    self.data.update(parsed_data)
TypeError: 'NoneType' object is not iterable

cyrils, 2023-08-22:
Should be fixed now, can you check again?

pedrong1025, 2023-08-22:
HI Cyrils, unfortunately, it still doesn't work this time. Here is the output.

INFO:root:[FC:A8:9B:20:6A:1B] Connected
INFO:root:[FC:A8:9B:20:6A:1B] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 5000 => [48, 3, 19, 136, 0, 17, 5, 73]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation30032200040024002400240024000000000000000000000000000000000000000000000000c1e4
DEBUG:root:create_request_payload 5017 => [48, 3, 19, 153, 0, 18, 21, 77]
INFO:root:characteristic_write_value_succeeded
WARNING:root:on_data_received: unknown operation=131

cyrils, 2023-08-22:
Ah.. ok. Does that work if we change  `cell_volt_info` words to 18 and `cell_temp_info` words to 17 (`BatteryClient#21-22`). Sorry its taking some iterations.

pedrong1025, 2023-08-22:
hi Cyrils, it looks good this time. Below is the output for both controller and battery. One suggestion to make the code better is to check if the output is good, e.g. no connection failed...etc. If not, make a loop to run the code again. Just my 2 cents :)

Battery
DEBUG:root:BT-TH-9B2xxxx => {'function': 'READ', 'cell_count': 4, 'cell_voltage_0': 3.3000000000000003, 'cell_voltage_1': 3.4000000000000004, 'cell_voltage_2': 3.3000000000000003, 'cell_voltage_3': 3.3000000000000003, 'sensor_count': 4, 'temperature_0': 24.0, 'temperature_1': 24.0, 'temperature_2': 24.0, 'temperature_3': 24.0, 'current': -2.7800000000000002, 'voltage': 13.5, 'remaining_charge': 99.53, 'capacity': 100.0, 'model': 'RBT100LFP12S-G'}

Control
DEBUG:root:BT-TH-EA8xxxxx => {'function': 'READ', 'model': '  RNG-CTRL-RVR', 'battery_percentage': 100, 'battery_voltage': 13.3, 'battery_current': 2.2, 'battery_temperature': 25, 'controller_temperature': 33, 'load_status': 'on', 'load_voltage': 13.3, 'load_current': 0.24, 'load_power': 3, 'pv_voltage': 72.60000000000001, 'pv_current': 0.41000000000000003, 'pv_power': 30, 'max_charging_power_today': 315, 'max_discharging_power_today': 6, 'charging_amp_hours_today': 38, 'discharging_amp_hours_today': 5, 'power_generation_today': 506, 'power_consumption_today': 67, 'power_generation_total': 5489, 'charging_status': 'mppt', 'battery_type': 'lithium'}

closed 2023-08-23 by cyrils