cyrils/renogy-bt #21: Test Renogy 48v battery compatibility

cyrils, 2023-08-23:
Test compatibility of Renogy Battery [RBT50LFP48S](https://www.renogy.com/support/downloads/batteries/lithium-batteries/48v-50ah-smart-lithium-iron-phosphate-battery-rbt50lfp48s.html)

jimmystewpot, 2023-08-26:
I have one of these batteries, I am currently working on getting it to work.

cyrils, 2023-08-26:
Sure, let me know how that goes. Here is the register mapping if you need [Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf](https://github.com/cyrils/renogy-bt/files/12444500/Lithium.Iron.Battery.BMS.Modbus.Protocol.V1.7.zh-CN.en.1.pdf)

cyrils, 2023-08-31:
@jimmystewpot Any luck?

jimmystewpot, 2023-08-31:
@cyrils no success  yet, but I am working on it when I have spare time.

cyrils, 2023-09-01:
Sure. If you can share the hex output may be I can also help.  In BaseClient#61
```py
logging.info(f"on_data_received: response for read operation{response.hex()}")
```

closed 2024-11-20 by cyrils