cyrils/renogy-bt #22: Charge controller history

tbarmann, 2023-08-26:
I just set this up using the Renogy Wanderer/ Raspberry Pi Zero W and it's working perfectly. Thank you!

Is there a way to get the history data that the Renogy app can retrieve?

cyrils, 2023-08-26:
I believe its definitely possible because the app does it. However I could not make sense of the data in the past. Let me know if anyone figure this out. Documentation isn't very clear either [Rover 20A/40A Charge Controller—MODBUS Protocol](https://docs.google.com/document/d/1OSW3gluYNK8d_gSz4Bk89LMQ4ZrzjQY6/edit) (search for "historical data")

cyrils, 2023-08-27:
Ok, here is an experimental branch for reading [historical-data](https://github.com/cyrils/renogy-bt/tree/historical-data), can you check it out and let me know if it works for you?

You have to change the device type in config.ini to `RNG_CTRL_HIST`.

tbarmann, 2023-08-27:
Wow, thanks Cyrils!

It seems to be working, though I'm not sure how to interpret the values. Here's the output:
```
INFO:root:Init RoverHistoryClient: BT-TH-161DADB1 => ac:4d:16:1d:ad:b1
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 6
INFO:root:Found matching device BT-TH-161DADB1     => [ac:4d:16:1d:ad:b1]
INFO:root:[ac:4d:16:1d:ad:b1] Connected
INFO:root:[ac:4d:16:1d:ad:b1] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 61446 => [255, 3, 240, 6, 0, 10, 3, 18]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61445 => [255, 3, 240, 5, 0, 10, 243, 18]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61444 => [255, 3, 240, 4, 0, 10, 162, 210]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61443 => [255, 3, 240, 3, 0, 10, 19, 19]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61442 => [255, 3, 240, 2, 0, 10, 66, 211]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61441 => [255, 3, 240, 1, 0, 10, 178, 211]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 61440 => [255, 3, 240, 0, 0, 10, 227, 19]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-161DADB1 => {'function': 'READ', 'power_generation': [0, 0, 27, 40, 0, 38, 13], 'amp_hours': [0, 0, 2, 3, 0, 3, 1], 'max_power': [0, 5, 5, 6, 1, 9, 8], '__device': 'BT-TH-161DADB1', '__client': 'RoverHistoryClient'}
INFO:root:Exit: Disconnecting device: BT-TH-161DADB1 [ac:4d:16:1d:ad:b1]

```

cyrils, 2023-08-27:
Thats daily stats for last 7 days in chronological order (last one is today's data).

closed 2023-10-03 by cyrils