cyrils/renogy-bt #24: installing directly in HA

Abudinka, 2023-08-28:
I am running a Raspi4 B with native installation of HA. 

It seems like there is no easy way to build libscrc on that machine. I was wondering if anybody could supply a binary for that or one could get rid of the dependency, so we could run this script on native HA?

cyrils, 2023-08-29:
I've removed libscrc dependency.

Abudinka, 2023-08-29:
well now iam dealing with missing dbus, probably gonna look at Issue #12 and google a bit.

cyrils, 2023-09-09:
Were you able to resolve the issue? May be installing `dbus-python` will help
``

Abudinka, 2023-09-10:
unfortunately no, terminal says dbus is installed, but wont find it when running the script. 

when i finally got dbus out of the way, its complaining about `gi` instead

i dont know if the script some kind of raspberry exclusive, but couldnt make gatt work with ubuntu either

marcuslib, 2023-09-10:
I got it to work in the end by installing HA in docker and then had to install python within a VM. To run it i then had to create a service to run the python script, it will also check if its running and if not run it, as I did find it took a few times to run the script for it to pickup the BT-1.

Abudinka, 2023-09-11:
> I got it to work in the end by installing HA in docker and then had to install python within a VM. To run it i then had to create a service to run the python script, it will also check if its running and if not run it, as I did find it took a few times to run the script for it to pickup the BT-1.

yes, i noticed. But i dont want to move HA back to docker. I used to use docker, but couldnt use addons, so i switched to native. 

I dont really understand why gatt is behaving the way it does. maybe we can dockerize the whole thing to make it easier? i tried something along the lines, but ran into the very same issues. plus you need some way to expose bt to the docker. 

I will most likely try a different ble interface and clone the renogy logic, dont know tbh

didi767, 2023-10-26:
@Abudinka is there a guide on how to implement this in HA? I'm not an expert, but I was planning to install Ubuntu and then HA on my orange pi zero 3.
Im sure it would be an adventure to figure this out, but if you have any clear steps on how to achieve it that would be awesome.
At first i thought about going with the RS485 method until I saw that we can actually interpret the BL2, which is amazing!

Abudinka, 2023-10-27:
sorry cant help here, as i said i dropped attempting renogy bluetooth communication since there is nothing to gain for me. if available i recommend using the home assistant raspberry pi image directly instead of using ubuntu. initially i used ubuntu and HA docker, but it has some drawbacks like no addon support. switched to raspberry pi 4B and used the HA image available in [Raspberry Pi Imager](https://www.raspberrypi.com/software/).

cruscio, 2023-11-26:
Ran into `Module 'dbus' not found` on headless Raspberry PI. Something in all of this fixed it:

```
pip3 install --user -r requirements.txt
sudo apt-get install python3-dbus
sudo apt install libdbus-1-3 libdbus-1-dev
sudo apt-get install libglib2.0-dev
sudo apt install python3-pydbus
python3 -m pip install --user dbus-python
```

And then I ran into
```
  File "/home/cruscio/.local/lib/python3.12/site-packages/gatt/gatt_linux.py", line 13, in <module>
    from gi.repository import GObject
ModuleNotFoundError: No module named 'gi'
```

Which should be fixed by running
```
pip3 install --user PyGObject
```

But that fails ... So then:
```
sudo apt install libcairo2 libcairo2-dev
pip3 install --user PyGObject
```

*Another* new failure. Log says to install `libgirepository1.0-dev`
```
sudo apt install libgirepository1.0-dev
pip3 install --user PyGObject
```

Finally! The `PyGObject` install worked, though it might have been easier to install libgtk2.0-dev (which a [stackoverflow post for libglib2.0-dev mentioned](https://askubuntu.com/questions/58321/how-do-i-install-glib)

And now I get 
```
ModuleNotFoundError: No module named 'requests'
```

./facepalm 
```
pip3 install --user requests
```

ANNNNND ***finally*** ... 
```
python3 ./example.py config.ini
INFO:root:Init BatteryClient: BT-TH-XXXXXXXX => XX:XX:XX:XX:XX:XX
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
```

Couple thoughts:
1. @cyrils - It seems a few requirements are missing from `requirements.txt` (not complaining. I definitely appreciate all the work you've shared here. Just wanted to call out that it may be worth adding *all* the requirements to `requirements.txt` :) )
2. I don't understand why there are so many things (glib2, cairo, GObject) that seem tied to GTK in a non-gui application...? (maybe this has something to do with accessing BT?) 

Now if only I could figure out how to discover my battery info without needing to dismantle my battery box to muck with the daisy chain. (Bluetooth logging while running the android app shows chatter in wireshark, but I'm not smart enough to decipher it :( )

Anyway - I hope this helps others with similar issues.

cyrils, 2023-11-26:
Hm.. I think the `gatt` library uses these for some reason. A regular raspberry pi image includes all of these by default, never tested it over headless image. Doesn't hurt to include them into requirements.txt though.

And unfortunately the app does not show the device id. Currently the only way to get the individual Ids is to connect one device at a time.  However you can try some usual suspects for batteries:
```
33, 34, 35.. series
48, 49, 50..
97, 98..
247, 248..
```

Abudinka, 2024-07-24:
update: 

came back to the this project out of curiosity, made it work on ubuntu earlier today, still some prerequisites, but its working now, ill post my history, trying to highlight the most relevant commands, **this is works on my machine stuff** 

```455  python3 -m venv .vevn
  456  source .vevn/bin/activate
  457  pip install -r requirements.txt
  458  sudo apt install dbus-python
  459  apt search dbus-python
  460  pip install dbus-python
  461  gcc -version
  462  sudo apt install gcc
  463  pip install -r requirements.txt
  464  apt search dbus
  465  sudo apt install python3-dbus
  466  sudo apt install dbus
  467  pip install -r requirements.txt
  468  sudo apt install pkg-config cmake
  469  pip install -r requirements.txt
  470  sudo apt install dbus-1
  471  sudo apt install build-essential libpython3-dev libdbus-1-dev
  472  pip install -r requirements.txt
  473  sudo apt install glib-2.0
  474  apt search glib
  475  sudo apt install libglib2.0-0
  476  sudo apt-get install libglib2.0-dev
  477  pip install -r requirements.txt
  478  pip install pycairo
  479  dpkg -h
  480  dpkg --help
  481  sudo apt install cairo
  482  sudo apt install python3-cairo
  483  ^[[200~Package 'cairo', required by 'virtual:world', not found
  484  ~Package 'cairo', required by 'virtual:world', not found\n
  485  sudo apt-get --no-install-recommends install pkg-config libcairo2-dev libpython3-dev gcc
  486  ~Package 'cairo', required by 'virtual:world', not found\n
  487  pip install -r requirements.txt
  488  apt search gobject
  489  sudo apt install libgirepository1.0-dev 
  490  pip install -r requirements.txt
  491  l
  492  python3 ./example.py config.ini
```

milperks, 2024-12-03:
> update:
> 
> came back to the this project out of curiosity, made it work on ubuntu earlier today, still some prerequisites, but its working now, ill post my history, trying to highlight the most relevant commands, **this is works "on my machine" stuff** , look for <--
> 
> ```
>   456  source .vevn/bin/activate
>   457  pip install -r requirements.txt
>   458  sudo apt install dbus-python
>   459  apt search dbus-python
>   460  pip install dbus-python
>   461  gcc -version
>   462  sudo apt install gcc  <--
>   463  pip install -r requirements.txt
>   464  apt search dbus
>   465  sudo apt install python3-dbus
>   466  sudo apt install dbus
>   467  pip install -r requirements.txt
>   468  sudo apt install pkg-config cmake <--
>   469  pip install -r requirements.txt
>   470  sudo apt install dbus-1
>   471  sudo apt install build-essential libpython3-dev libdbus-1-dev  <--
>   472  pip install -r requirements.txt
>   473  sudo apt install glib-2.0
>   474  apt search glib
>   475  sudo apt install libglib2.0-0
>   476  sudo apt-get install libglib2.0-dev <--
>   477  pip install -r requirements.txt
>   478  pip install pycairo
>   479  dpkg -h
>   480  dpkg --help
>   481  sudo apt install cairo
>   482  sudo apt install python3-cairo
>   483  ^[[200~Package 'cairo', required by 'virtual:world', not found
>   484  ~Package 'cairo', required by 'virtual:world', not found\n
>   485  sudo apt-get --no-install-recommends install pkg-config libcairo2-dev libpython3-dev gcc <--
>   486  ~Package 'cairo', required by 'virtual:world', not found\n
>   487  pip install -r requirements.txt
>   488  apt search gobject
>   489  sudo apt install libgirepository1.0-dev  <--
>   490  pip install -r requirements.txt
>   491  l
>   492  python3 ./example.py config.ini
> ```
> 
> trying to get it run on HA OS now, different kind of beast cause it does not come with `apt` but `apk` which i am not familiar with at all

Were you able to progress this any further? There is someone on the HA forums who has made a custom integration for this but I couldn't get that working for me (https://github.com/markgdev/home-assistant_renogy)

cyrils, 2024-12-04:
@milperks dependencies have changed a lot. Not sure if this issue still exists.