cyrils/renogy-bt #26: Python error when running example.py in crontab

tbarmann, 2023-09-07:
I can run the example.py program from the command line on a Raspberry Pi Zero W. I have modified the program to log data to a Google spreadsheet. Running it from the command line works great.

But when I try running it with crontab, it doesn't seem to run. On a subsequent run from the command line, I get this error:
```
INFO:root:on_data_received: response for read operation
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/gatt/gatt_linux.py", line 539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=bytes(value))
  File "/home/pi/Documents/solar/renogy-bt/renogybt/BLE.py", line 86, in characteristic_value_updated
    self.data_callback(value)
  File "/home/pi/Documents/solar/renogy-bt/renogybt/RoverClient.py", line 59, in on_data_received
    super().on_data_received(response)
  File "/home/pi/Documents/solar/renogy-bt/renogybt/BaseClient.py", line 62, in on_data_received
    index, section = self.find_section_by_response(response)
TypeError: cannot unpack non-iterable NoneType object
```
My crontab entry:
```
*/5 * * * * python3 /home/pi/Documents/solar/renogy-bt/example.py
```

Any idea why it won't work with crontab but it does work from the command line?

cyrils, 2023-09-07:
You seem to be on an older codebase, its hard to debug. Can you update from master and let me know the output?

tbarmann, 2023-09-07:
Using the latest master, I created a fork and added my code for google sheets logging: https://github.com/tbarmann/renogy-bt/tree/add-gsheet-logging
You are welcome to include it if you like. It probably needs some instructions in the readme, which I can add if you are interested.

It works great from the command line, but it doesn't execute using crontab. It no longer creates an error on subsequent runs from the command line.

Here is the console output for the successful run from the command line, if that helps:
```
pi@raspberrypi:~/Documents/renogy-bt $ python3 /home/pi/Documents/renogy-bt/example.py
INFO:root:Init RoverClient: BT-TH-161DADB1 => ac:4d:16:1d:ad:b1
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 8
INFO:root:Found matching device BT-TH-XXXXXXXX    => [XX:XX:XX:XX:XX:XX]
INFO:root:[XX:XX:XX:XX:XX:XX] Connected
INFO:root:[XX:XX:XX:XX:XX:XX] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 12 => [255, 3, 0, 12, 0, 8, 145, 209]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 57348 => [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-161DADB1 => {'function': 'READ', 'model': 'RNG-CTRL-WND10', 'battery_percentage': 98, 'battery_voltage': 13.100000000000001, 'battery_current': 0.4, 'battery_temperature': 0, 'controller_temperature': 39, 'load_status': 'on', 'load_voltage': 13.100000000000001, 'load_current': 0.24, 'load_power': 3, 'pv_voltage': 14.0, 'pv_current': 0.35000000000000003, 'pv_power': 5, 'max_charging_power_today': 5, 'max_discharging_power_today': 3, 'charging_amp_hours_today': 1, 'discharging_amp_hours_today': 4, 'power_generation_today': 13, 'power_consumption_today': 13, 'power_generation_total': 628, 'charging_status': 'boost', 'battery_type': 'sealed', '__device': 'BT-TH-161DADB1', '__client': 'RoverClient'}
DEBUG:urllib3.util.retry:Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
DEBUG:urllib3.connectionpool:https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): sheets.googleapis.com:443
DEBUG:urllib3.connectionpool:https://sheets.googleapis.com:443 "GET /v4/spreadsheets/XXXXXXXXXXXXXXXXXXXXXX?includeGridData=false HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): www.googleapis.com:443
DEBUG:urllib3.connectionpool:https://www.googleapis.com:443 "GET /drive/v3/files/XXXXXXXXXXXXXXXXXXXXXX?supportsAllDrives=True&includeItemsFromAllDrives=True&fields=id%2Cname%2CcreatedTime%2CmodifiedTime HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:https://sheets.googleapis.com:443 "GET /v4/spreadsheets/XXXXXXXXXXXXXXXXXX?includeGridData=false HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:https://sheets.googleapis.com:443 "GET /v4/spreadsheets/XXXXXXXXXXXXXXXXXXXXXXX/values/%27Sheet1%27%21A1%3A1 HTTP/1.1" 200 None
DEBUG:urllib3.connectionpool:https://sheets.googleapis.com:443 "POST /v4/spreadsheets/XXXXXXXXXXXXXXXXXXXXXXX/values/%27Sheet1%27:append?valueInputOption=RAW&includeValuesInResponse=False HTTP/1.1" 200 None
INFO:root:Exit: Disconnecting device: BT-TH-XXXXXXXX [XX:XX:XX:XX:XX:XX]
```
(I've X'd out private info).

cyrils, 2023-09-07:
Can you share the new error output while running from crontab?

tbarmann, 2023-09-07:
I found the issue. It was not with your code. My google sheet logging branch requires a credentials file. I needed to use the full path to that file in the config.ini file when triggering the program from crontab. (A relative path is sufficient for running on the command line).

 It seems to be working great now! Thanks so much for your responsiveness Cyrils.

Creating a log file helped solve this:
```
*/5 * * * * python3 /home/pi/Documents/renogy-bt/example.py >> /home/pi/Documents/renogy-bt/renogy.log 2>&1
```

closed 2023-09-07 by cyrils