cyrils/renogy-bt #28: Not consistently reading values

tbarmann, 2023-09-10:
I occasionally see the app not reading the values from the BT-1. The data shows only values that do not change, ie `function`, `model`, etc.  Here is the log: 

```
pi@raspberrypi:~/Documents/renogy-bt $ python3 example.py
INFO:root:Init RoverClient: BT-TH-161DADB1 => ac:4d:16:1d:ad:b1
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 1
INFO:root:Found matching device BT-TH-161DADB1     => [ac:4d:16:1d:ad:b1]
INFO:root:[ac:4d:16:1d:ad:b1] Connected
INFO:root:[ac:4d:16:1d:ad:b1] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 12 => [255, 3, 0, 12, 0, 8, 145, 209]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 57348 => [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:characteristic_write_value_failed
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-161DADB1 => {'function': 'READ', 'model': 'RNG-CTRL-WND10', '__device': 'BT-TH-161DADB1', '__client': 'RoverClient'}
```

This happened four times in a row over the course of an hour, running at 15 minute intervals. Other times, it works fine. Could this be a hardware problem? Or an async issue?

cyrils, 2023-09-10:
I can see a `characteristic_write_value_failed`, probably due to connectivity issue. How far is the device from your pi?

tbarmann, 2023-09-10:
It's 6 inches away.

tbarmann, 2023-09-11:
I added that logging to the BaseClient. When it fails, it doesn't hit it. Here's the log.
```
INFO:root:Init RoverClient: BT-TH-161DADB1 => ac:4d:16:1d:ad:b1
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 1
INFO:root:Found matching device BT-TH-161DADB1 => [ac:4d:16:1d:ad:b1]
INFO:root:[ac:4d:16:1d:ad:b1] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 12 => [255, 3, 0, 12, 0, 8, 145, 209]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 57348 => [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 57348 => [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:characteristic_write_value_failed
INFO:root:characteristic_write_value_failed
INFO:root:characteristic_write_value_failed
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-161DADB1 => {'function': 'READ', 'model': 'RNG-CTRL-WND10', '__device': 'BT-TH-161DADB1', '__client': 'RoverClient'}
```

cyrils, 2023-09-11:
Can you add a `time.sleep(1)` at  BaseClient#74 before the `self.read_section()`?

tbarmann, 2023-09-11:
I added that line. I'll let you know if it makes a difference. Thanks.

cyrils, 2023-09-11:
Good to hear, can you try lowering that to sleep(0.2)? Just want to know whats the lower limit.

tbarmann, 2023-09-11:
After 78 successful reads, there was one failure.

cyrils, 2023-09-11:
Ok, did you get a chance with to test with lower values? The issue is most probably the back to back writes for different sections. Just trying to avoid the 1 sec delay as there can be as many as 7 sections for batteries.

tbarmann, 2023-09-12:
Results from testing with `sleep(0.2)`: 80 successful reads, 8 failed reads, over 21 hours.

cyrils, 2023-09-13:
May be 0.2 is too small. I'll change it to 0.5 as an interim fix. Might need to make a refactor later for handling exception properly. Or you can also create a PR if you have have changes ready.

cyrils, 2023-09-16:
Fixed in master

closed 2023-09-16 by cyrils