cyrils/renogy-bt #30: Load and power date from Rover controller is all zeros

calebu2, 2023-09-24:
Successfully got this working on a pi zero with a smart battery (RBT100LFP12SH) and rover 40 mppt connected via separate bt1 and bt2 modules.

Battery stats are fine and some of the rover stats are good too, but many of the controller stats are zero:

I will see if I can debug the bytestream and report back, but for now here is the parsed payload:

DEBUG:root:BT-TH-Exxxxxxx => {'function': 'READ', 'model': 'RNG-CTRL-RVR', 'battery_percentage': 99, 'battery_voltage': 13.100000000000001, 'battery_current': 0.0, 'battery_temperature': 19, 'controller_temperature': 21, 'load_status': 'on', 'load_voltage': 13.100000000000001, 'load_current': 0.0, 'load_power': 0, 'pv_voltage': 14.0, 'pv_current': 0.0, 'pv_power': 0, 'max_charging_power_today': 6, 'max_discharging_power_today': 26, 'charging_amp_hours_today': 0, 'discharging_amp_hours_today': 1, 'power_generation_today': 0, 'power_consumption_today': 13, 'power_generation_total': 160652, 'charging_status': 'mppt', 'battery_type': 'lithium', '__device': 'BT-TH-Exxxxxxx', '__client': 'RoverClient'}
INFO:root:mqtt logging
INFO:root:Exit: Disconnecting device: BT-TH-Exxxxxxx [c4:xx:xx:xx:xx:xx]

cyrils, 2023-09-24:
To debug change BaseClient#61
```py
logging.info(f"on_data_received: response for read operation{response.hex()}")
```

calebu2, 2023-09-25:
Here's the bytes for the controller data (2nd of the three strings returned):
ff0344005b0082000013120082000000000004000000000001008100850039000000060000000000000000000002d88705012c00002fdf00001f680002738c00019342800000001ea8

Not sure what to make of it.

cyrils, 2023-09-25:
Those values are in fact zero. This is the value Rover returned.
```py
{'function': 'READ', 'battery_percentage': 91, 'battery_voltage': 13.0, 'battery_current': 0.0, 'battery_temperature': 18, 'controller_temperature': 19, 'load_status': 'on', 'load_voltage': 13.0, 'load_current': 0.0, 'load_power': 0, 'pv_voltage': 0.4, 'pv_current': 0.0, 'pv_power': 0, 'max_charging_power_today': 6, 'max_discharging_power_today': 0, 'charging_amp_hours_today': 0, 'discharging_amp_hours_today': 0, 'power_generation_today': 0, 'power_consumption_today': 0, 'power_generation_total': 160652, 'charging_status': 'deactivated'}
```

closed 2023-09-26 by cyrils

calebu2, 2023-09-26:
Thanks for creating this package. Apologies for the confusion.

Realizing that people in a similar circumstance may find this closed issue I will share my diagnostic process as my penance:

1) double check with the renogy app (either renogy bt or dc home app) that you are not also getting zeros in the app.
2) if zeros in app, double check the live current/power display on the controller.
3) reboot the controller (I did not have the manual so I disconnected it from battery and solar)

Doing these steps should restore the data. If that still doesnt work, controller may be defective.