cyrils/renogy-bt #39: Error when calling set_load() function

CStar01, 2023-11-04:
I'm trying to write a simple script to toggle the charge controller's load on and off.  
client.set_load(1) works as expected and the controller's load turns on
but client.set_load(0) throws an error and does not turn off the load
---Error Log--
INFO:root:on_read_operation_complete
Data received
INFO:root:setting load 0
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 232, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 539, in properties_changed
    self.service.device.characteristic_value_updated(characteristic=self, value=bytes(value))
  File "/home/pi/renogy-bt/renogybt/BLE.py", line 86, in characteristic_value_updated
    self.data_callback(value)
  File "/home/pi/renogy-bt/renogybt/RoverClient.py", line 57, in on_data_received
    super().on_data_received(response)
  File "/home/pi/renogy-bt/renogybt/BaseClient.py", line 74, in on_data_received
    self.on_read_operation_complete()
  File "/home/pi/renogy-bt/renogybt/BaseClient.py", line 88, in on_read_operation_complete
    self.on_data_callback(self, self.data)
  File "/home/pi/renogy-bt/toggle.py", line 24, in on_data_received
    client.set_load(0)
  File "/home/pi/renogy-bt/renogybt/RoverClient.py", line 67, in set_load
    self.device.characteristic_write_value(request)
  File "/home/pi/renogy-bt/renogybt/BLE.py", line 89, in characteristic_write_value
    self.write_characteristic.write_value(value)
  File "/home/pi/.local/lib/python3.9/site-packages/gatt/gatt_linux.py", line 565, in write_value
    bytes = [dbus.Byte(b) for b in value]
TypeError: 'NoneType' object is not iterable

--My code--
def on_data_received(client, data):
  print("Data received")
  loadswitch = 0
  if data["load_status"] == "off": loadswitch = 1
  client.set_load(loadswitch)
  client.disconnect()

#Connect
RoverClient(config, on_data_received).connect()

cyrils, 2023-11-04:
I have merged a fix, can you check?

CStar01, 2023-11-04:
It is working now.  Thank you
I will add a function check, thanks again.

closed 2023-11-04 by CStar01