cyrils/renogy-bt #41: Test Inverter compatibility

cyrils, 2023-11-06:
Use branch [inverter](https://github.com/cyrils/renogy-bt/tree/inverter) to test compatibility with Renogy inverter models:

- [ ] RIV4835CSH1S-US
- [ ] RNG-INVT-XX series
- [ ] R-INVT-PCL1-XX series

The `InverterClient` is experimental and may be incomplete/ incompatible with some models.

braathwaate, 2023-12-31:
This is running the Lycan 5000 using the InverterClient.  Is this the output that you want to see?

python3 example.py
INFO:root:Init InverterClient: BT-TH-F26D8F08 => F0:F8:F2:6D:8F:08
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 1
INFO:root:Found matching device BT-TH-F26D8F08     => [f0:f8:f2:6d:8f:08]
INFO:root:[F0:F8:F2:6D:8F:08] Connected
INFO:root:[F0:F8:F2:6D:8F:08] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 4000 => [32, 3, 15, 160, 0, 8, 65, 139]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_stats 20031004d7000004b000141771021c01090000bfc1
DEBUG:root:create_request_payload 4311 => [32, 3, 16, 215, 0, 8, 246, 69]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_model 20031052495634383335435348315300000000162c
DEBUG:root:create_request_payload 4329 => [32, 3, 16, 233, 0, 5, 86, 76]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_solar_charging 20030a03ab000300200002002030b0
DEBUG:root:create_request_payload 4410 => [32, 3, 17, 58, 0, 2, 231, 139]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_load 200304001700007b35
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-F26D8F08 => {'function': 'READ', 'uei_voltage': 1239, 'uei_current': 0, 'voltage': 1200, 'load_current': 20, 'frequency': 6001, 'temperature': 540, 'model': 'RIV4835CSH1S\x00\x00', 'solar_voltage': 939, 'solar_current': 3, 'solar_power': 32, 'solar_charging_state': 2, 'solar_charging_power': 32, 'load_power': 23, 'charging_current': 0, '__device': 'BT-TH-F26D8F08', '__client': 'InverterClient'}
INFO:root:Exit: Disconnecting device: BT-TH-F26D8F08 [F0:F8:F2:6D:8F:08]

cyrils, 2023-12-31:
Thanks! Are these values correct? May be there is a scaling issue so solar_voltage=93.9, load_current=0.2A etc?

braathwaate, 2023-12-31:
This is the actual output.  I've fixed some of the scaling issues in the LycanClient in my fork.  And  battery_current is wacky and I do not understand it.  Here is the output from LycanClient:
DEBUG:root:BT-TH-F26D8F08 => {'function': 'READ', 'uei_voltage': 123.6, 'uei_current': 0, 'voltage': 120.0, 'load_current': 20, 'frequency': 59.94, 'battery_temperature': 82.4, 'controller_temperature': 35.6, 'model': 'RIV4835CSH1S\x00\x00', 'solar_voltage': 926, 'solar_current': 78, 'solar_power': 729, 'solar_charging_state': 2, 'solar_charging_power': 729, 'load_power': 23, 'charging_current': 0, 'battery_percentage': 79, 'battery_voltage': 54.0, 'battery_current': 654.01, 'pv_voltage': 92.4, 'pv_current': 8.0, 'pv_power': 745, 'battery_type': 'custom', '__device': 'BT-TH-F26D8F08', '__client': 'LycanClient'}
INFO:root:Exit: Disconnecting device: BT-TH-F26D8F08 [F0:F8:F2:6D:8F:08]

cyrils, 2024-01-01:
I've merged a basic support to master, but still not fully tested as I don't have a device.

braathwaate, 2024-01-02:
Here is the output:

INFO:root:Init InverterClient: BT-TH-F26D8F08 => F0:F8:F2:6D:8F:08
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 1
INFO:root:Found matching device BT-TH-F26D8F08     => [f0:f8:f2:6d:8f:08]
INFO:root:[F0:F8:F2:6D:8F:08] Connected
INFO:root:[F0:F8:F2:6D:8F:08] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 4000 => [32, 3, 15, 160, 0, 8, 65, 139]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:characteristic_value_updated 21
INFO:root:on_data_received: response for read operation 21
INFO:root:parse_inverter_stats 20031004d3000004af000a177201f5010900003052
DEBUG:root:create_request_payload 4311 => [32, 3, 16, 215, 0, 8, 246, 69]
INFO:root:characteristic_write_value_succeeded
INFO:root:characteristic_value_updated 21
INFO:root:on_data_received: response for read operation 21
INFO:root:parse_inverter_model 20031052495634383335435348315300000000162c
DEBUG:root:create_request_payload 4329 => [32, 3, 16, 233, 0, 5, 86, 76]
INFO:root:characteristic_write_value_succeeded
INFO:root:characteristic_value_updated 15
INFO:root:on_data_received: response for read operation 15
INFO:root:parse_solar_charging 20030a03670022012c0001012ce4f2
DEBUG:root:create_request_payload 4410 => [32, 3, 17, 58, 0, 2, 231, 139]
INFO:root:characteristic_write_value_succeeded
INFO:root:characteristic_value_updated 9
INFO:root:on_data_received: response for read operation 9
INFO:root:parse_inverter_load 200304000c00000b32
DEBUG:root:create_request_payload 57348 => [32, 3, 224, 4, 0, 1, 244, 186]
INFO:root:characteristic_write_value_succeeded
INFO:root:characteristic_value_updated 7
INFO:root:on_data_received: response for read operation 7
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-F26D8F08 => {'function': 'READ', 'uei_voltage': 123.5, 'uei_current': 0.0, 'voltage': 119.9, 'load_current': 10, 'frequency': 60.02, 'temperature': 50.1, 'model': 'RIV4835CSH1S', 'solar_voltage': 87.1, 'solar_current': 3.4, 'solar_power': 300, 'solar_charging_state': 'activated', 'solar_charging_power': 300, 'load_power': 12, 'charging_current': 0.0, 'battery_type': 'custom', '__device': 'BT-TH-F26D8F08', '__client': 'InverterClient'}
INFO:root:Exit: Disconnecting device: BT-TH-F26D8F08 [F0:F8:F2:6D:8F:08]

mavenius, 2024-07-11:
I have a R-INVT-PCL1-30111S-US, and I'm curious how you are expecting to connect to it. It is my understanding that it netiher has bluetooth, nor is its RJ45 port connected to anything internal. The RJ11 is connected for two status LEDs (power and error) and remote power switch.

I'd be thrilled to hear that I could get some data out of it.

cyrils, 2024-07-18:
Yes, looks like R-INVT-PCL1 series don't support it.

ab0oo, 2024-08-04:
I can confirm this is working for my RIV4835CSH1S:

```
$ python3 ./example.py config.ini 
INFO:root:Init InverterClient: BT-TH-EA83778C => e0:7d:ea:83:77:8c
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 6
INFO:root:Found matching device BT-TH-EA83778C     => [e0:7d:ea:83:77:8c]
INFO:root:[e0:7d:ea:83:77:8c] Connected
INFO:root:[e0:7d:ea:83:77:8c] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 4000 => [32, 3, 15, 160, 0, 8, 65, 139]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_stats 20031000000000000000000000021c018600006257
DEBUG:root:create_request_payload 4311 => [32, 3, 16, 215, 0, 8, 246, 69]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_model 20031052495634383335435348315300000000162c
DEBUG:root:create_request_payload 4329 => [32, 3, 16, 233, 0, 5, 86, 76]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_solar_charging 20030a03e2000100100002001008f3
DEBUG:root:create_request_payload 4410 => [32, 3, 17, 58, 0, 2, 231, 139]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_load 20030400000000cb31
DEBUG:root:create_request_payload 57348 => [32, 3, 224, 4, 0, 1, 244, 186]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-EA83778C => {'function': 'READ', 'uei_voltage': 0.0, 'uei_current': 0.0, 'voltage': 0.0, 'load_current': 0, 'frequency': 0.0, 'temperature': 54.0, 'model': 'RIV4835CSH1S', 'solar_voltage': 99.4, 'solar_current': 0.1, 'solar_power': 16, 'solar_charging_state': 'mppt', 'solar_charging_power': 16, 'load_power': 0, 'charging_current': 0.0, 'battery_type': 'custom', '__device': 'BT-TH-EA83778C', '__client': 'InverterClient'}
```

The secret sauce for me was the "32" as byte[0], which I'm assuming is the modbus address.  I have a BT2 with both the RIV4835CSH1S and a RBT50LFP48S "Smart" battery plugged into it, using MODBUS.
Perhaps update the matrix in the README to show address 32 for the INVERTER address?

Also, what's the config file format to read both devices with a single script?  A second [device] block?

cyrils, 2024-08-05:
Each device needs a separate config file. May be you can create a new script file for reading the two devices back to back.
```sh
python3 ./example.py config1.ini
python3 ./example.py config2.ini
```

team1401, 2024-11-15:
I have the Renogy Inverter RIV4835CSH1S and everything seems to work and present the proper data except for the battery info. I'm using non-Renogy Lithium batteries, and it IDs them as "None". I don't really care if it IDs them properly, but would it be possible to pull the battery metrics from the inverter? I'd love to be able to log Battery Charging Watts, Volts, Charging Amps, and Discharging Amps.

For reference, this is what I get running it against my inverter:

`python3 ./example.py config.ini
INFO:root:Init InverterClient: BT-TH-481B0E18 => F8:55:48:1B:0E:18
INFO:root:Starting discovery...
INFO:root:Devices found: 34
INFO:root:Found matching device BT-TH-481B0E18     => F8:55:48:1B:0E:18
INFO:root:Client connection: True
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 15, 160, 0, 8, 82, 228]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_stats ff031004d5013604d9013617710248018d0000aaec
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 16, 215, 0, 8, 229, 42]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_model ff031052495634383335435348315300000000b9eb
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 16, 233, 0, 5, 69, 35]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
INFO:root:on_data_received: response for read operation
INFO:root:parse_solar_charging ff030a00000000000000020000d309
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 17, 58, 0, 2, 244, 228]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
INFO:root:on_data_received: response for read operation
INFO:root:parse_inverter_load ff03040167000055df
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
INFO:root:BT-TH-481B0E18     => {'function': 'READ', 'uei_voltage': 123.7, 'uei_current': 31.0, 'voltage': 124.1, 'load_current': 310, 'frequency': 60.01, 'temperature': 58.4,                                    'model': 'RIV4835CSH1S', 'solar_voltage': 0.0, 'solar_current': 0.0, 'solar_power': 0, 'solar_charging_state': 'mppt', 'solar_charging_power': 0, 'load_power': 359, 'charging_c                                   urrent': 0.0, 'battery_type': None, '__device': 'BT-TH-481B0E18', '__client': 'InverterClient'}
INFO:root:mqtt logging
INFO:root:Exit: Disconnecting device: BT-TH-481B0E18     F8:55:48:1B:0E:18
`

cyrils, 2024-11-15:
Can you try reading the values as CONTROLLER and let me know the values?

cyrils, 2024-11-16:
I see, thats a first. Its actually possible to read battery data from inverter, I'lll work with you sometime next week to test a new branch.

cyrils, 2024-11-20:
Can you try https://github.com/cyrils/renogy-bt/tree/inverter-battery ?

team1401, 2024-11-20:
I haven't been able to get that one to run cleanly. A bunch of errors that were solved by items not in the dependency list, but now this is what I get when I run it:

brian@nas:~/renogy-bt-inverter-battery $ python ./example.py config.ini
Traceback (most recent call last):
  File "/home/brian/renogy-bt-inverter-battery/./example.py", line 5, in <module>
    from renogybt import InverterClient, RoverClient, RoverHistoryClient, BatteryClient, DataLogger, Utils
  File "/home/brian/renogy-bt-inverter-battery/renogybt/__init__.py", line 1, in <module>
    from .RoverClient import RoverClient
  File "/home/brian/renogy-bt-inverter-battery/renogybt/RoverClient.py", line 2, in <module>
    from .BaseClient import BaseClient
  File "/home/brian/renogy-bt-inverter-battery/renogybt/BaseClient.py", line 7, in <module>
    from .BLE import DeviceManager, Device
  File "/home/brian/renogy-bt-inverter-battery/renogybt/BLE.py", line 1, in <module>
    import gatt
  File "/usr/local/lib/python3.11/dist-packages/gatt/__init__.py", line 1, in <module>
    from .gatt import DeviceManager, Device, Service, Characteristic
  File "/usr/local/lib/python3.11/dist-packages/gatt/gatt.py", line 6, in <module>
    from .gatt_linux import *
  File "/usr/local/lib/python3.11/dist-packages/gatt/gatt_linux.py", line 13, in <module>
    from gi.repository import GObject
  File "/home/brian/.local/lib/python3.11/site-packages/gi/__init__.py", line 39
    print url
    ^^^^^^^^^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?

team1401, 2024-11-20:
I'm using the bleak port: https://github.com/cyrils/renogy-bt/pull/66

cyrils, 2024-11-24:
Main branch is now updated with better inverter support, with more fields from official documentation.