cyrils/renogy-bt #42: Renogy Core One support coming?

didi767, 2023-11-08:
Would it be possible to read directly from the Core One?
My mppt and battery are connected to it, but I've noticed that I can't tap to the BT-2 and sending data to my mqtt server.
Has anyone noticed that or thought about how to resolve that? (not sure if it's considered as an issue or maybe enhancement request?)

Regards,
Didi

cyrils, 2023-11-08:
I'm not aware of any Core One APIs. Since bt-2 is already hooked into core one, no other clients can connect to it.

didi767, 2023-11-08:
Can I use the script with the MAC address of the Core One instead of the BT-2?
Currently my MPPT is using the BT-2 to connect to the core one, and my battery is connected over the internal/built-in BT.
So I was thinking if I could just use the Core one with the integration to read data from the two components

closed 2024-11-20 by cyrils