cyrils/renogy-bt #45: BT-2 device is undiscoverable

cruscio, 2023-11-26:
Using android Bluetooth HCI snooping while running the Renogy DC Home app, I can see traffic to my BT-2 device in Wireshark and I can see battery info in the app. However the device never appears in the bluetooth menus of my phone, mac laptop, windows laptop, or in bluetoothctl `scan on` on a raspberry pi. 

When I run example.py with the alias/mac info from the wireshark trace, it appears to cycle through the discoverable devices in the area and then report `device not found`

I'm having trouble understanding how this should work if the BT-2 device isn't discoverable by design. Am I missing a firmware update or setting on the BT-2 to make it discoverable?

```
python3 ./example.py config.ini
INFO:root:Init BatteryClient: BT-TH-774C10AA => f4:60:****
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 1
INFO:root:Checking device Ice Box => [34:94:****]
INFO:root:Devices found: 1
INFO:root:Checking device Ice Box => [34:94:****]
INFO:root:Devices found: 1
INFO:root:Checking device Ice Box => [34:94:****]
INFO:root:Devices found: 1
INFO:root:Checking device Ice Box => [34:94:****]
INFO:root:Devices found: 1
INFO:root:Checking device Ice Box => [34:94:****]
ERROR:root:Device not found: BT-TH-774C10AA => f4:60:****, please check the details provided.
```

In the above log, _Ice Box_ is my fridge. It never finds the BT-2 device even though it's on, working, and a couple feet away (within a similar range as the phone app). 

That `Checking` log line comes from a modification to `BLE.py` to see what it was looping though:
```
        while discovering:
            time.sleep(1)
            logging.info("Devices found: %s", len(self.devices()))
            for dev in self.devices():
                logging.info("Checking device %s => [%s]", dev.alias(), dev.mac_address)
                if dev.mac_address != None and (dev.mac_address.upper() == mac_address or dev.alias() == self.device_alias) and discovering:
                    logging.info("Found matching device %s => [%s]", dev.alias(), dev.mac_address)
                    discovering = False; self.device_found = True
```

I've tried running both example.py and bluetoothctl scanning at different times: during & shortly after running the app (in case something special needs to wake it up), and also tried after the app hasn't been connected (in case it'll only talk to one device at a time). The BT-2 device never appears.

cyrils, 2023-11-26:
BT-2 is definitely discoverable, thats how BLE works in the first place. However it can connect to only one device at a time, so kill you mobile app and restart your pi to make sure there is no thread hanging in somewhere.

Also the BLE advertising happens at a fixed interval so two consecutive scans might fail. Wait for at least 15 seconds between scans.

cruscio, 2023-11-26:
Success! Turned the phone off, rebooted the Pi, and it worked. After getting mixed results with `device_id = 255` (expected per #34), I got lucky on the first try with `33` & `34` from your suggestion in #24

If it'd ever be useful to have someone logging/testing something on a BT-2 and two daisy-chained batteries, tag me in the issue and I'd be glad to help.

Thank you!

closed 2023-11-26 by cruscio