cyrils/renogy-bt #46: Error when using raspberry pi 4

dawson-friesenhahn, 2023-11-27:
I've been using your repo on a very old raspberry pi model B v 2.0 running  Linux raspberry pi 6.1.21+ with great success. I had to use a USB dongle for Bluetooth, but it worked without a hitch. 

I recently upgraded to a raspberry pi model 4B running Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2, and I cant for the life of me get connected. The error message comes from gatt I believe, and isn't very helpful:

`(.venv) pi@raspberrypi:~/renogy-bt $ python example.py
INFO:root:Init RoverClient: BT-TH-774B0D88 => f4:60:77:4b:0d:88
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 16
INFO:root:Found matching device BT-TH-774B0D88     => [f4:60:77:4b:0d:88]
ERROR:root:Connection failed: le-connection-abort-by-local`

I've tried using both the pi 4's native Bluetooth hardware and the old functioning Bluetooth dongle with similar results. bluetooth, bluez, and gatt are all the most recent versions. 

Any advice in terms of avenues I should pursue for further debugging?

cyrils, 2023-11-27:
I haven't really tested it on Pi4 hardware. What does your `etc/bluetooth/main.conf` say? Have you tried using an older version of bluez (like 5.5)?

emresaglam, 2024-01-17:
I created a docker container for renogy-bt for `linux/arm/v8` and I'm using it with my Rpi4. If interested I can throw a quick write up. It might help you @dawson-friesenhahn

dawson-friesenhahn, 2024-01-17:
That would be fantastic. I've put this project on the back burner for now, but that write-up could prove very useful once I'm able to pick it up again. Thanks!

dawson-friesenhahn, 2024-01-17:
Also, my /etc/bluetooth/main.conf:

```
[General]
# Defaults to 'BlueZ X.YZ', if Name is not set here and plugin 'hostname' is not loaded.
# The plugin 'hostname' is loaded by default and overides the Name set here so
# consider modifying /etc/machine-info with variable PRETTY_HOSTNAME=<NewName> instead.
#Name = BlueZ
# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100
# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
#DiscoverableTimeout = 0
# Always allow pairing even if there are no agent registered
# Possible values: true, false
# Default: false
#AlwaysPairable = false
# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
#PairableTimeout = 0
# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd
# Do reverse service discovery for previously unknown devices that connect to
# us. For BR/EDR this option is really only needed for qualification since the
# BITE tester doesn't like us doing reverse SDP for some test cases, for LE
# this disables the GATT client functionally so it can be used in system which
# can only operate as peripheral.
# Defaults to 'true'.
#ReverseServiceDiscovery = true
# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true
# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false
# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = dual
# Enables Multi Profile Specification support. This allows to specify if
# system supports only Multiple Profiles Single Device (MPSD) configuration
# or both Multiple Profiles Single Device (MPSD) and Multiple Profiles Multiple
# Devices (MPMD) configurations.
# Possible values: "off", "single", "multiple"
#MultiProfile = off
# Permanently enables the Fast Connectable setting for adapters that
# support it. When enabled other devices can connect faster to us,
# however the tradeoff is increased power consumptions. This feature
# will fully work only on kernel version 4.1 and newer. Defaults to
# 'false'.
#FastConnectable = false
# Default privacy setting.
# Enables use of private address.
# Possible values: "off", "device", "network"
# "network" option not supported currently
# Defaults to "off"
# Privacy = off
# Specify the policy to the JUST-WORKS repairing initiated by peer
# Possible values: "never", "confirm", "always"
# Defaults to "never"
#JustWorksRepairing = never
# How long to keep temporary devices around
# The value is in seconds. Default is 30.
[... 75 lines cut ...]
# Number of ATT channels
# Possible values: 1-5 (1 disables EATT)
# Default to 3
#Channels = 3
[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb,0000111f-0000-1000-8000-00805f9b34fb,0000110a-0000-1000-8000-00805f9b34fb
# ReconnectAttempts define the number of attempts to reconnect after a link
# lost. Setting the value to 0 disables reconnecting feature.
#ReconnectAttempts=7
# ReconnectIntervals define the set of intervals in seconds to use in between
# attempts.
# If the number of attempts defined in ReconnectAttempts is bigger than the
# set of intervals the last interval is repeated until the last attempt.
#ReconnectIntervals=1,2,4,8,16,32,64
# AutoEnable defines option to enable all controllers when they are found.
# This includes adapters present on start as well as adapters that are plugged
# in later on. Defaults to 'false'.
AutoEnable=true
```

matthewcobb, 2024-01-18:
On a Pi5, experiencing the same error. `ERROR:root:Connection failed: le-connection-abort-by-local` - The only thing I would add is that occasionally it does connect; if you retry the script 30-40 times or such... Once connected it is quite stable however.

matthewcobb, 2024-01-23:
Just an update, I rewrote the bluetooth management side of this for my own purposes to connect using Bleak. https://github.com/hbldh/bleak and it is now working great on Raspberry Pi 5. It also seems to provide a much simpler way of managing device connections and callbacks.

cyrils, 2024-01-23:
@matthewcobb Feel free to raise a PR against the `bleak` branch.

cyrils, 2024-04-03:
Experimental [bleak](https://github.com/cyrils/renogy-bt/pull/66) branch is now available

closed 2024-11-21 by cyrils