cyrils/renogy-bt #51: dump example.py to text file or json locally?

bserret, 2023-12-12:
I would like to just dump the output to a text file, or a local json file but I see no way to do that - a redirect "python example.py >>output.txt" does not work. The php post seems to work (code 200) but I am not skilled enough with php to have it dump  to a file - how does it even execute on the server side? I presume $json_data is getting populated but a file_put_contents doesn't do anything. Likely my misunderstanding of how php works regarding dumping output to a local file on the server...

Thanks for any suggestions.

cyrils, 2023-12-13:
Turn off all logging in your `config.ini` and write your code after `example.py#L18`.

```py
with open('dump.txt', 'w') as f:
    f.write(json.dumps(filtered_data))
```

bserret, 2023-12-13:
I had to add import json but otherwise that worked perfectly thanks!

closed 2023-12-15 by cyrils