cyrils/renogy-bt #55: ML2440

FrancineMatos, 2024-01-10:
Just to confirm it's working on ML2440 with BT-1
[image]

FrancineMatos, 2024-01-10:
Just to clarify a thing i have to add 2 spaces after the alias of the bluethoot device in the config.ini to work.

Like this: "BT-TH-73C8EA91  "

FrancineMatos, 2024-01-10:
Everything workin wiuth MQTT integrated to HA.

[image]

cyrils, 2024-01-11:
Great! I'll add it to the doc.

> Just to clarify a thing i have to add 2 spaces after the alias of the bluethoot device in the config.ini to work.

Thats strange as it matches either the mac adress or alias. Are you sure your mac addr is correct in the config?

FrancineMatos, 2024-01-11:
> Great! I'll add it to the doc.
> 
> > Just to clarify a thing i have to add 2 spaces after the alias of the bluethoot device in the config.ini to work.
> 
> Thats strange as it matches either the mac adress or alias. Are you sure your mac addr is correct in the config?

Yes If try to connect without the 2 spaces it's says that the device was not found even If the Mac is correct. I belive that the device broadcast the alias with those spaces.

cyrils, 2024-01-13:
I pushed a change, hopefully it fixed the alias issue.

FrancineMatos, 2024-01-15:
> I pushed a change, hopefully it fixed the alias issue.

Thanks, now the device was found without the spaces in the alias. Even looks like that it's imprrove the connection with the device, some times it took a while to connect and start to send information, but now it's connecting very fast.

closed 2024-01-15 by cyrils