cyrils/renogy-bt #60: Temperature showing up as hotter than surface of the sun.

Anglac, 2024-02-12:
My battery temperatures have been reporting at 11,000+ degrees f, given the lack of fires I believe this to be false.

I think the issue is that the battery temperature returns a signed int, and it's being handled as if it's unsigned.

switching line 37 of BatteryClient.py to 

            celcius = bytes_to_int(bs, 5 + i*2, 2, scale = 0.1, signed=True)

seems to have it reporting more reasonable temperatures.

cyrils, 2024-02-16:
@Anglac What are the values you see when adding signed flag?

Anglac, 2024-02-16:
Started getting below freezing (sub 32f or 0c) temperatures after adding the signed flag.
(below freezing is expected values in my area in the winter time)

Before the change was getting things like 11826.68f or 6552.6c (which was suspiciously close to 65,535 limit for unsigned 16 bit int when scaled by 0.1).

cyrils, 2024-02-17:
Fixed in master

closed 2024-02-17 by cyrils