cyrils/renogy-bt #61: Renogy Smart Shunt 300 Bluetooth

tvan0076, 2024-03-04:
My Renogy Rover 60A works like a champ! I thought I would try to use the same code for the Renogy Shunt 300 with Bluetooth, but that didn't work so well :-(  It doesn't like the request payload.   

DEBUG:root:create_request_payload 5000 => [33, 3, 19, 136, 0, 17, 6, 8]

I used the RNG_BATT in hopes they were close, but I guess not. I tried many device_IDs but couldn't get it to read.

Output Data:
INFO:root:Init BatteryClient: RTMShunt300xxxxxxxx => xx:xx:xx:xx:xx:xx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 55
INFO:root:Found matching device RTMShunt3005xxxxxxxx => [(xx:xx:xx:xx:xx:xx)]
INFO:root:[4c:e1:74:4c:92:98] Connected
INFO:root:[4c:e1:74:4c:92:98] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 5000 => [33, 3, 19, 136, 0, 17, 6, 8]
INFO:root:characteristic_enable_notifications_failed
INFO:root:characteristic_write_value_succeeded
ERROR:root:on_read_timeout => please check your device_id!
INFO:root:Exit: Disconnecting device: RTMShunt300xxxxxxxx [(xx:xx:xx:xx:xx:xx)]

Is there a way to get past this easily and pull the data?

cyrils, 2024-03-05:
Looks like Shunt works differently, characteristic_enable_notifications_failed is not a good sign. Is this the same response you get everytime?

Only way you can figure out the register values is by doing bluetooth snooping. You have to enable Bluetooth HCI snoop in android phone while using dc home app and open the file using wireshark ([read](https://sps-support.honeywell.com/s/article/How-to-capture-Bluetooth-traffic-from-and-to-an-Android-Device)).

tvan0076, 2024-03-05:
I have created the snoop file with iOS and XCode.  Would you be open to looking at it to see what might need to be done to add the shunt to your program? If so, how should I send it.  It has too much private info to add to the thread.

tvan0076, 2024-03-05:
Here is the ATT Attributes:
[image]

cyrils, 2024-03-06:
Upload the file to google drive and send me the link to `hejikob514@mcuma.com`

cyrils, 2024-03-06:
How do I open the file?

tvan0076, 2024-03-06:
I opened it with wireshark.Thanks,Terry VanCleve972.533.4780

cyrils, 2024-03-06:
Also send a screenshot of the app.

tvan0076, 2024-03-06:
Here you go.Thanks,Terry VanCleve972.533.4780

cyrils, 2024-03-07:
I think the file is enough. I didn't get the screenshot btw. It's better if log file and screenshot are taken together.

tvan0076, 2024-03-10:
Let me know if I need to test any modifications :) happy to be a tester as I might be the only one (so far) that has one. Much appreciated!

cyrils, 2024-03-13:
I briefly looked into it. Unfortunately the data sent back from Shunt looks very different from rest of the devices. There are some patterns but it will be really difficult to figure out the whole story without an actual device.

tvan0076, 2024-03-13:
At this point I would be happy to ship you the shunt if you are willing to make the integration. I really need this for automating when I am not at home. I don’t want my battery bank (400 ah) below 50%.  I use MQTT to talk to Home Assistant and Node Red to auto turn on the battery charger when the shunt shows less than 55% capacity. I don’t want to lose the meat in the refrigerator/freezer while not at home. This is easy when at home, but when traveling this is a much bigger problem. Thoughts are welcome.  I am script kiddy, and I am not a developer :(.  Thanks,Terry VanCleve972.533.4780

cyrils, 2024-03-13:
@tvan0076 your replies don't show up in git for some reason.

Unfortunately I'm going for a month long vacation this week, so even if you ship it the timing is not correct. In the mean time can't you use battery soc from renogy for automation? Its voltage based, so not accurate like shunt but you can experiment with different values and see if it works.

cyrils, 2024-03-13:
If you can send a new set of screenshot/logs taken at the same time I can take a look at it when I'm away. Currently I cannot map the log to actual values as the screenshot seems to be taken at a different time.

tvan0076, 2024-03-13:
I sent a zip file that has new log with screen shots to you a few days ago. Did you receive the email?Thanks,Terry VanCleve972.533.4780

tvan0076, 2024-03-13:
Attached is the new logs with screen shots.

Begin forwarded message:

> From: Terry Van Cleve ***@***.***>
> Date: March 7, 2024 at 10:13:31 AM CST
> To: ***@***.***
> Subject: New log and screen shots
> 
> ﻿
> Please see attached zip file.  I took new logs and screenshots at the same time. I hope this helps.  Also I did change the battery AHs to 400 and the Sync battery to 92% so that should reflect in the logs.

cyrils, 2024-03-13:
That email id has expired. can you forward it to ***@***.***?

tvan0076, 2024-03-13:
The email address is blanked out.***@***.***?

tvan0076, 2024-03-13:
Try this one:Thanks,Terry VanCleve972.533.4780

cyrils, 2024-03-13:
wites37618@fryshare.com

cyrils, 2024-03-22:
Can you let me know what do you see when using this branch [shunt](https://github.com/cyrils/renogy-bt/tree/shunt)? 

Use device type `RNG_SHUNT`

tvan0076, 2024-03-22:
The first output was this:
INFO:root:Init SmartShuntClient: RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 21
INFO:root:Found matching device RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:[xx:xx:xx:xx:xx:xx] Connected
INFO:root:[xx:xx:xx:xx:xx:xx] Resolved services
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 218, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/gatt/gatt_linux.py", line 398, in properties_changed
    self.services_resolved()
  File "/home/pi/renogy-bt-main/renogybt/BLE.py", line 73, in services_resolved
    self.resolved_callback()
  File "/home/pi/renogy-bt-main/renogybt/BaseClient.py", line 58, in __on_resolved
    self.poll_data() if self.config['data'].getboolean('enable_polling') == True else self.read_section()
  File "/home/pi/renogy-bt-main/renogybt/BaseClient.py", line 105, in read_section
    request = self.create_generic_read_request(self.device_id, 3, self.sections[index]['register'], self.sections[index]['words'])
TypeError: create_generic_read_request() takes 1 positional argument but 5 were given
INFO:root:characteristic_enable_notifications_failed

After the first output I kept getting this:
INFO:root:Init SmartShuntClient: RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 21
INFO:root:Found matching device RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:[xx:xx:xx:xx:xx:xx] Connected
INFO:root:[xx:xx:xx:xx:xx:xx] Resolved services
ERROR:root:Exception occured: create_generic_read_request() takes 1 positional argument but 5 were given

I hope this helps.

cyrils, 2024-03-22:
Updated, please check again.
Doo you always get this error `characteristic_enable_notifications_failed`?

tvan0076, 2024-03-22:
The test above I didn't get the characteristic_enable_notifications_failed.  I don't think it made it that far.

Updated test I get this:
INFO:root:Init SmartShuntClient: RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 15
INFO:root:Found matching device RTMShunt30053001400 => [xx:xx:xx:xx:xx:xx]
INFO:root:[xx:xx:xx:xx:xx:xx] Connected
INFO:root:[xx:xx:xx:xx:xx:xx] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
INFO:root:characteristic_enable_notifications_failed
INFO:root:characteristic_write_value_succeeded
ERROR:root:on_read_timeout => please check your device_id!
INFO:root:Exit: Disconnecting device: RTMShunt30053001400 [xx:xx:xx:xx:xx:xx]

cyrils, 2024-03-22:
That means the shunt has totally different ble characteristics. You might have to use a ble scanner app to figure out which one is write characteristic and which one is notification and update in BaseClient

tvan0076, 2024-03-22:
Here is the output for both the controller and the shunt.  The shunt doesn't use the subscribed to notification like the controller.
    Device (new): xx.xx.xx.xx.xx.xx (public), -36 dBm
        Flags: <06>
        Incomplete 16b Services: <0000ffd0-0000-1000-8000-00805f9b34fb>
        Manufacturer: <6cb2fd859e6a>
        Tx Power: <00>
        Complete Local Name: 'BT-TH-FD859E6A    '
    Device (new): xx.xx.xx.xx.xx.xx (public), -30 dBm
        Flags: <06>
        Manufacturer: <4ce1744c9298001e0001>
        Appearance: <0000>
        Tx Power: <00>
        Complete Local Name: 'RTMShunt30053001400'

tvan0076, 2024-03-22:
I tried running sensortag --all xx.xx.xx.xx.xx.xx and I got this back:
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral xx.xx.xx.xx.xx.xx, addr type: public

tvan0076, 2024-03-22:
sudo gatttool -b xx:xx:xx:xx:xx:xx -I
[xx:xx:xx:xx:xx:xx][LE]> connect
Attempting to connect to xx.xx.xx.xx.xx.xx
Error: connect: Connection refused (111)

tvan0076, 2024-03-22:
I sent pic of the BLE scanner to the email address above. Lots of good data :-)

tvan0076, 2024-04-08:
I hope you are having an amazing holiday!!! FYI deleting this post to GitHub.Thanks,Terry VanCleve972.533.4780

cyrils, 2024-08-05:
Closing it as unsupported.

closed 2024-08-05 by cyrils

morpheusrogue, 2024-08-06:
Just ran into this same problem with my new setup..i had hope that you were hot on the trail, and JUST 2 days ago, this closed :( 
I'm COMPLETELY new to this BLE interrogation/interpretation, but i need (want??) to get at this data for Grafana or other (better!) logging.  Maybe this info helps? 

I was able to use LightBlue BLE scanner on the Mac and this seems to be the notification stream 
Name: RTMShunt30017000621
Device UUID: 696A6318-82C6-B188-E78C-071F1182CE5A
Service UUID: 0x0000C011-0000-1000-8000-00805F9B34FB 
Descriptor: 0x2902 

Content example stream notification: 
```10:16:06.223 - Characteristic (0000C411-0000-1000-8000-00805F9B34FB) notified: <42570119 01001630 a0240806 a1230836 a202137f a4ffddf1 a60033e5 a6100000 a70003c5 a7100000 a80bc6b0 a9000001 aa04e800 f8010000 ad01010e ad0200c9 ad0300c9 ad048000 b1210000 b1230000 b1240000 b3210000 b3230000 b3240000 b5210000 b5230000 b5240000 db04> 
10:16:06.342 - Characteristic (0000C411-0000-1000-8000-00805F9B34FB) notified: <42570119 01001630 a0240806 a1230836 a202137f a4ffddf1 a60033e5 a6100000 a70003c5 a7100000 a80bc6b0 a9000001 aa04e800 f8010000 ad01010e ad0200c9 ad0300c9 ad048000 b1210000 b1230000 b1240000 b3210000 b3230000 b3240000 b5210000 b5230000 b5240000 db04> 
10:16:06.463 - Characteristic (0000C411-0000-1000-8000-00805F9B34FB) notified: <42570119 01001630 a0240806 a1230836 a202137f a4ffddf1 a60033e5 a6100000 a70003c5 a7100000 a80bc6b0 a9000001 aa04e800 f8010000 ad01010e ad0200c9 ad0300c9 ad048000 b1210000 b1230000 b1240000 b3210000 b3230000 b3240000 b5210000 b5230000 b5240000 db04> 
10:16:06.581 - Characteristic (0000C411-0000-1000-8000-00805F9B34FB) notified: <42570119 01001630 a0240806 a1230836 a202137f a4ffddf1 a60033e5 a6100000 a70003c5 a7100000 a80bc6b0 a9000001 aa04e800 f8010000 ad01010e ad0200c9 ad0300c9 ad048000 b1210000 b1230000 b1240000 b3210000 b3230000 b3240000 b5210000 b5230000 b5240000 db04> ```

cyrils, 2024-08-07:
The smart shunt follows a different protocol, its hard to decode without an actual device.