cyrils/renogy-bt #62: MQTT Discovery support

FrancineMatos, 2024-03-05:
I
[Files.zip](https://github.com/cyrils/renogy-bt/files/14501324/Files.zip)
've added the MQTT Discovery function that auto create the components in Home Assistant.

1 - We need to add this to DataLogger.py (for some reason i cant paste the code here it's broke the editor)
[image]

2 - some changes need to be done on log_mqtt function still in DataLogger.py
[image]

3 - We need to create the call for the new function on example.py or whathever you are using to start the renogy
[image]

A video fo the thing working:
https://github.com/cyrils/renogy-bt/assets/148077119/13f61b24-5d9e-4b15-a75d-4f838eb8766e

FrancineMatos, 2024-03-05:
Another improvment, now it's return the device class and the unit of measurement of the components by Home Assistant MQTT Discovery so the sensor is correctly registred.

Was necessary to change a bit the data structure but i dont think that it can do something bad

Some changes in DataLogger.py and RoverClient.py

[Files.zip](https://github.com/cyrils/renogy-bt/files/14502256/Files.zip)

cyrils, 2024-03-07:
Instead of the zip file can you create a PR against branch [mqtt-discovery](https://github.com/cyrils/renogy-bt/tree/mqtt-discovery)?

FrancineMatos, 2024-03-07:
I'm not famillyar with githut plataform but i cant try. I ll find some tutorial teaching how to do it.

FrancineMatos, 2024-03-07:
> Instead of the zip file can you create a PR against branch [mqtt-discovery](https://github.com/cyrils/renogy-bt/tree/mqtt-discovery)?

Ok, i think i did it, my first time using github to update a repository but i guess i did it rigth.

closed 2024-03-08 by cyrils

KodinLanewave, 2024-08-13:
This was marked as completed however never merged.  Do you mind digging back into that branch and merging at some point?

KodinLanewave, 2024-08-13:
Whoops, I tried that branch and found a syntax error in one of the files; part of one of the lines was clipped and a > put in instead.  I'll see if I can fork that, modify, and submit an MR for that branch to at least make the file workable so it can be merged successfully.

FrancineMatos, 2024-08-14:
> @cyrils @FrancineMatos Whoops, I tried that branch and found a syntax error in one of the files; part of one of the lines was clipped and a > put in instead. I'll see if I can fork that, modify, and submit an MR for that branch to at least make the file workable so it can be merged successfully.
> 
> Edit: Branched, Fixed, Merged, PR submitted. Code changes tested successfully, they populate just fine in HA automatically on my HA instance.

Hii, nice that worked for you, it's already branched in main->mqtt-discovery, i think there is working fine.