cyrils/renogy-bt #67: Help with RBT12100LFP-BT

Brakelmann, 2024-04-24:
Hi,

I tried to get my new Renogy RBT12100LFP-BT working, the example.py throws an error.

```
INFO:root:Init RoverClient: RNGPRO12BATT4600xxxx => 4c:e1:74:49:xx:xx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 5
INFO:root:Found matching device RNGPRO12BATT4600xxxx => [4c:e1:74:49:xx:xx]
INFO:root:[4c:e1:74:49:xx:xx] Connected
INFO:root:[4c:e1:74:49:xx:x] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 12 => [255, 3, 0, 12, 0, 8, 145, 209]
INFO:root:characteristic_enable_notifications_failed
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_failed
WARNING:root:on_data_received: unknown operation=131
^CERROR:root:Exception occured: KeyboardInterrupt
INFO:root:Exit: Disconnecting device: RNGPRO12BATT46000121 [4c:e1:74:49:xx:xx]
```

Any help is appreciated.
Thanks, Stefan.

cyrils, 2024-04-25:
Looks like you use PRO series battery and it has different register values. It will be impossible to figure out the actual values without bluetooth snooping using something like wireshark.

Brakelmann, 2024-04-25:
Ok, thanks for you response.
Renogy has an Android App to connect via BT and read the values.

Would it somhow be possible to sniff the communication between this app and the battery's BMS to get the register values?
Eg with some BT Proxy running on the PiZero?

cyrils, 2024-04-25:
Yes you have to use the android app and something like wireshark to capture the traffic to a log file.

Brakelmann, 2024-04-25:
Before I dig deeper into this one more question.
Would it be sufficient to use the internal Bluetooth of the RPI Zero 2W to sniff with wireshark, or do I need an external BT Stick?
e.g.: https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/quickstart-guide

Thanks, Stefan.

cyrils, 2024-04-25:
You shouldn't be using RPI. Read the traffic from your iOS/Android phone when using the renogy home app. Method differs based on your mobile platform. Ex: [Android](https://sps-support.honeywell.com/s/article/How-to-capture-Bluetooth-traffic-from-and-to-an-Android-Device)

Brakelmann, 2024-04-25:
OK, I got it.
Thanks for pointing me there.

Brakelmann, 2024-04-25:
I was able to capture the communication between Phone and Battery.
Downloded the capture file and opened it with wireshark.
I was also able to set a filter based on the mac adresses of the phone and battery.

What would be the next steps?

[image]

cyrils, 2024-04-25:
Can you send the log file to kobel98713@togito.com?

Brakelmann, 2024-04-25:
Sent the file to you some minutes ago...

cyrils, 2024-04-28:
Hm.. newer renogy devices seems to be using different protocols. Its hard to figure out without owning a device.

Brakelmann, 2024-04-28:
If I can log or test something for you just let me know... maybe we can figure that out working together.

cyrils, 2024-08-05:
Closing as unsupported

closed 2024-08-05 by cyrils