cyrils/renogy-bt #68: How to run 2 config files ?

conor-o-brien, 2024-04-24:
Hi, Is it possible to run 2 config files at same time and if so, how?
I can run them both individually config1.ini and config.ini
The reason I have to do this is because I am using both Bt1 and Bt2 devices.

cyrils, 2024-04-25:
Not sure what you meant by at the same time. You will need separate config file for each device, but you can run them back to back or at the same time. Ex: create a sh file with
```sh
python3 example.py config1.ini 
python3 example.py config2.ini 
```

conor-o-brien, 2024-04-28:
Thanks

> Not sure what you meant by at the same time. You will need separate config file for each device, but you can run them back to back or at the same time. Ex: create a sh file with
> 
> ```shell
> python3 example.py config1.ini 
> python3 example.py config2.ini 
> ```

Thanks, I got that working when adding to crontab.

conor-o-brien, 2024-04-28:
> Not sure what you meant by at the same time. You will need separate config file for each device, but you can run them back to back or at the same time. Ex: create a sh file with
> 
> ```shell
> python3 example.py config1.ini 
> python3 example.py config2.ini 
> ```

Thanks, I got that working when adding to crontab

closed 2024-04-28 by conor-o-brien