cyrils/renogy-bt #78: gatt.errors.Failed: Did not receive a reply.

RaoulSargent, 2024-08-19:
Running Raspberry Pi OS on Raspberry Pi Zero W, and getting the following error, hoping someone can help please.

I retrieved the BT-1 Bluetooth Alias and MAC address using nrfConnect on iOS.
Renogy Wanderer with single Battery.

```
python ./example.py config.ini
INFO:root:Init RoverClient: BT-TH-16178A45 => 4D:AC:16:17:8A:45
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 34
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/gatt/gatt_linux.py", line 376, in alias
    return self._properties.Get('org.bluez.Device1', 'Alias')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 634, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/raoul/renogy/renogy-4ts/./example.py", line 34, in <module>
    RoverClient(config, on_data_received, on_error).connect()
  File "/home/raoul/renogy/renogy-4ts/renogybt/BaseClient.py", line 33, in connect
    self.manager.discover()
  File "/home/raoul/renogy/renogy-4ts/renogybt/BLE.py", line 29, in discover
    if dev.mac_address != None and (dev.mac_address.upper() == mac_address or (dev.alias() and dev.alias().strip() == self.device_alias)) and discovering:
                                                                                               ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/gatt/gatt_linux.py", line 384, in alias
    raise _error_from_dbus_error(e)
gatt.errors.Failed: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
```

cyrils, 2024-08-19:
Try moving closer to the device.

RaoulSargent, 2024-08-27:
Sorry for the delayed response.
I did try within 3m of the BT-1 using both an ESP32 (renogy-bt-esphome) and a Raspberry PI Zero W2 (renogy-bt), neither gets connected.
The Renogy iOS app can connect quite happily however.
Am I confused or doing something wrong?

RaoulSargent, 2024-08-27:
Making progress...

Using nrfConnect, I connected to my BT-TH-16178A45 and started to "Read" values using the "Down Arrow" against each property.
One of the properties is called "Hardware Revision String" with UUID: 2A27
When I pulled this value it gave me "AC4D16178A45", which looked like a MAC address to me.
Notice how the first 4 characters, in pairs are reversed... "AC4D" instead of "<4DAC>".
(In esphome I updated the ble_mac_address to read: AC:4D:16:17:8A:45 and am now getting connected)
So next I need to try again on the Raspberry PI with the new MAC address... will update here once done.

RaoulSargent, 2024-08-28:
Now using the "correct" MAC address it finds the BT-1 and WORKS :-)

```
$ python ./example.py config.ini
INFO:root:Init RoverClient: BT-TH-16178A45 => AC:4D:16:17:8A:45
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 26
INFO:root:Found matching device BT-TH-16178A45     => [ac:4d:16:17:8a:45]
INFO:root:[AC:4D:16:17:8A:45] Connected
INFO:root:[AC:4D:16:17:8A:45] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 12 => [255, 3, 0, 12, 0, 8, 145, 209]
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 26 => [255, 3, 0, 26, 0, 1, 176, 19]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 256 => [255, 3, 1, 0, 0, 34, 209, 241]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 57348 => [255, 3, 224, 4, 0, 1, 231, 213]
INFO:root:characteristic_write_value_succeeded
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:BT-TH-16178A45 => {'function': 'READ', 'model': 'RNG-CTRL-WND', 'device_id': 16, 'battery_percentage': 100, 'battery_voltage': 13.2, 'battery_current': 0.32, 'battery_temperature': 24, 'controller_temperature': 26, 'load_status': 'off', 'load_voltage': 0.0, 'load_current': 0.0, 'load_power': 0, 'pv_voltage': 13.2, 'pv_current': 0.32, 'pv_power': 4, 'max_charging_power_today': 5, 'max_discharging_power_today': 0, 'charging_amp_hours_today': 0, 'discharging_amp_hours_today': 0, 'power_generation_today': 0, 'power_consumption_today': 0, 'power_generation_total': 1332, 'charging_status': 'activated', 'battery_type': 'lithium', '__device': 'BT-TH-16178A45', '__client': 'RoverClient'}
INFO:root:mqtt logging
DEBUG:paho.mqtt.client:Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b'renogy-bt'
DEBUG:paho.mqtt.client:Received CONNACK (0, 0)
DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r0, m1), 'b'renogy/state'', ... (669 bytes)
DEBUG:paho.mqtt.client:Sending DISCONNECT
INFO:root:Exit: Disconnecting device: BT-TH-16178A45 [AC:4D:16:17:8A:45]
```

RaoulSargent, 2024-08-28:
@cyrils Thank you for creating this project.

closed 2024-08-29 by cyrils