cyrils/renogy-bt #80: Support for Renogy PRO 12V 200AH Smart Batteries

Sparhawk76, 2024-09-23:
I wanted to report that with some minor changes to BaseClient.py I was able to get the new Pro series batteries from Renogy working.  

Added:
ALIAS_PREFIX2 = 'RNGRBP'

and changed the line 
                if dev.alias() != None and dev.alias().startswith(ALIAS_PREFIX):
to:
                if dev.alias() != None and ( dev.alias().startswith(ALIAS_PREFIX) or dev.alias().startswith(ALIAS_PREFIX2) ):

There might be a better way to do this,  I'm quite rusty with my python skills, but it works.

Console output running the example script to do a scan:

INFO:root:Init RoverClient: BT-TH-B00FXXXX => 80:6f:b0:0f:xx:xx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 16
INFO:root:Devices found: 17
INFO:root:Devices found: 17
INFO:root:Devices found: 17
INFO:root:Devices found: 18
ERROR:root:Device not found: BT-TH-B00FXXXX => 80:6f:b0:0f:xx:xx, please check the details provided.
DEBUG:root:Possible device found! ======> RNGRBP1217000794 > [4c:e1:74:54:ca:dc]
DEBUG:root:Possible device found! ======> BT-TH-66EBCDDB > [60:98:66:eb:cd:db]
DEBUG:root:Possible device found! ======> BT-TH-EABD5E03 > [e0:7d:ea:bd:5e:03]
DEBUG:root:Possible device found! ======> BT-TH-66FA269B > [60:98:66:fa:26:9b]
DEBUG:root:Possible device found! ======> BT-TH-66FE6274 > [60:98:66:fe:62:74]
DEBUG:root:Possible device found! ======> BT-TH-66FB7BDC > [60:98:66:fb:7b:dc]
DEBUG:root:Possible device found! ======> BT-TH-481C0B29 > [f8:55:48:1c:0b:29]
DEBUG:root:Possible device found! ======> BT-TH-9B2F1DE7 > [fc:a8:9b:2f:1d:e7]
DEBUG:root:Possible device found! ======> BT-TH-FD85AAC8 > [6c:b2:fd:85:aa:c8]
DEBUG:root:Possible device found! ======> RNGRBP1217000792 > [4c:e1:74:4a:7d:36]
DEBUG:root:Possible device found! ======> RNGRBP1224000039 > [4c:e1:74:54:ca:78]

Colsole output from configured battery pasted below:

INFO:root:Init BatteryClient: RNGRBP1217000794 => 4c:e1:74:54:ca:dc
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 14
INFO:root:Found matching device RNGRBP1217000794 => [4c:e1:74:54:ca:dc]
INFO:root:[4c:e1:74:54:ca:dc] Connected
INFO:root:[4c:e1:74:54:ca:dc] Resolved services
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:resolved services
DEBUG:root:create_request_payload 5000 => [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_enable_notifications_failed
INFO:root:characteristic_enable_notifications_succeeded
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5017 => [255, 3, 19, 153, 0, 17, 68, 179]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5042 => [255, 3, 19, 178, 0, 6, 116, 181]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5122 => [255, 3, 20, 2, 0, 8, 245, 226]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5223 => [255, 3, 20, 103, 0, 1, 37, 251]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:RNGRBP1217000794 => {'function': 'READ', 'cell_count': 4, 'cell_voltage_0': 3.3, 'cell_voltage_1': 3.3, 'cell_voltage_2': 3.3, 'cell_voltage_3': 3.3, 'sensor_count': 3, 'temperature_0': 71.96, 'temperature_1': 71.78, 'temperature_2': 72.14, 'current': -0.3, 'voltage': 13.3, 'remaining_charge': 190.37, 'capacity': 217.07, 'model': 'RBT12200LFP-BT', 'device_id': 48, '__device': 'RNGRBP1217000794', '__client': 'BatteryClient'}
INFO:root:mqtt logging
DEBUG:root:create_request_payload 5000 => [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5017 => [255, 3, 19, 153, 0, 17, 68, 179]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5042 => [255, 3, 19, 178, 0, 6, 116, 181]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5122 => [255, 3, 20, 2, 0, 8, 245, 226]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
DEBUG:root:create_request_payload 5223 => [255, 3, 20, 103, 0, 1, 37, 251]
INFO:root:characteristic_write_value_failed
INFO:root:on_data_received: response for read operation
INFO:root:on_read_operation_complete
DEBUG:root:RNGRBP1217000794 => {'function': 'READ', 'cell_count': 4, 'cell_voltage_0': 3.3, 'cell_voltage_1': 3.3, 'cell_voltage_2': 3.3, 'cell_voltage_3': 3.3, 'sensor_count': 3, 'temperature_0': 71.96, 'temperature_1': 71.78, 'temperature_2': 72.14, 'current': -0.31, 'voltage': 13.3, 'remaining_charge': 190.37, 'capacity': 217.07, 'model': 'RBT12200LFP-BT', 'device_id': 48, '__device': 'RNGRBP1217000794', '__client': 'BatteryClient'}
INFO:root:mqtt logging

Sparhawk76, 2024-09-23:
One other change I didn't notice at first, but can compensate for in my node-red config is that current needs to be multiplied by 10 to get the correct output current.

cyrils, 2024-09-24:
Thats great to hear. But I don't think you need the above alias changes if the mac address is correct. Can you check with the correct mac (I assume `4c:e1:74:54:ca:dc`) if it works? 

The other batteries works too? What are their model numbers?

Sparhawk76, 2024-10-16:
Without the code, running the example.py to detect the batteries does not work. After a config file is made for each of the batteries, the added line is not needed. 

Running the example.py with the line changed back, there are no RNGBP batteries found, just my older BT-TH batteries, and my BT-TH solar controllers:
pi@openplotter:~/solar-monitor/renogy-bt1 $ python3 ./example.py
INFO:root:Init RoverClient: BT-TH-B00FXXXX => 80:6f:b0:0f:xx:xx
INFO:root:Adapter status - Powered: True
INFO:root:Starting discovery...
INFO:root:Devices found: 17
INFO:root:Devices found: 17
INFO:root:Devices found: 18
INFO:root:Devices found: 18
INFO:root:Devices found: 18
ERROR:root:Device not found: BT-TH-B00FXXXX => 80:6f:b0:0f:xx:xx, please check the details provided.
DEBUG:root:Possible device found! ======> BT-TH-66FE6274 > [60:98:66:fe:62:74]
DEBUG:root:Possible device found! ======> BT-TH-66EBCDDB > [60:98:66:eb:cd:db]
DEBUG:root:Possible device found! ======> BT-TH-EABD5E03 > [e0:7d:ea:bd:5e:03]
DEBUG:root:Possible device found! ======> BT-TH-66FA269B > [60:98:66:fa:26:9b]
DEBUG:root:Possible device found! ======> BT-TH-66FB7BDC > [60:98:66:fb:7b:dc]
DEBUG:root:Possible device found! ======> BT-TH-481C0B29 > [f8:55:48:1c:0b:29]
DEBUG:root:Possible device found! ======> BT-TH-9B2F1DE7 > [fc:a8:9b:2f:1d:e7]
DEBUG:root:Possible device found! ======> BT-TH-FD85AAC8 > [6c:b2:fd:85:aa:c8]

cyrils, 2024-10-19:
What I mean is if you give correct mac address it should connect without any code change. 80:6f:b0:0f.. doesn't match with any of your devices.

Sparhawk76, 2024-10-19:
IIRC I never changed the config.ini that is used by example.py, that mac address is the default in it.   So it shouldn't match with any of my devices.  I'm just using the example script to output the list of detected devices, and without my change this list will not have my new Renogy Pro batteries.

cyrils, 2024-10-19:
I got that part. But I mean you should change the mac address. Ex: 60:98:66:fe:62:74, then it should work without the code changes. Basically if either the alias OR mac_address matches it will connect.

Sparhawk76, 2024-10-19:
Yes, but at that point I have no need to connect, At that point I am just trying to find the mac addresses and the alias to put in my config files for each battery.   The code change allows that.   After I have added the battery to a config file and run a copy of example.py pointing to that battery's config, it will work with or without the code change as at that point I already know the mac address.

cyrils, 2024-10-20:
Gotcha. Can you share the battery models that works with this code?

Sparhawk76, 2024-10-20:
I've tested  this with the Renogy Pro 200AH W/BT and Heating, link pasted below.
[https://ca.renogy.com/pro-12v-200ah-smart-lithium-iron-phosphate-battery-w-bluetooth-self-heating/](url)
May also work with the 100AH model, but I have no way of testing that.

As I mentioned the amp reading reported by the batteries has to be multiplied by 10 (When 10 amps are flowing, it reads 1 amp without correction) to get the correct amperage flowing through the batteries, I've done this correction in Node-Red for my use case, but would be far nicer if this was corrected for in the renogy-bt code so doing the correction in Node-Red would not be needed.

cyrils, 2024-10-21:
Thanks. Its surprising though that they chose to change a single field in the pro series battery, but kept everything else intact. Are you sure it not something else?

Sparhawk76, 2024-10-21:
I have been monitoring the batteries and comparing to my shunt based monitor for the last few weeks ... with the 10 times modification the reported current matches what my shunt shows for that battery bank. Without it it show's 1/10th of what is flowing through the shunt. Seems odd that they'd change just one field, that's why I didn't catch it right away.

closed 2024-11-20 by cyrils