cyrils/renogy-bt #85: Installation instructions

jacksbox, 2024-10-26:
I had some trouble installing this repo (but maybe I missed something important?). 
I had to install several packages, before the `pip install` did run through:

```
sudo apt install libdbus-glib-1-dev libdbus-1-dev
sudo apt install libcairo2-dev libjpeg-dev libgif-dev
sudo apt install libgirepository1.0-dev
```
(raspberry pi zero 2w with Debian GNU/Linux 12 (bookworm))

I think this could be added to the README.md to make things simpler for other users.

cyrils, 2024-11-02:
Have you tried the [bleak](https://github.com/cyrils/renogy-bt/tree/bleak) branch?

cyrils, 2024-11-21:
Yes, I'l have to move main branch to bleak soon.

cyrils, 2024-11-21:
Master is now using bleak port, so the dependency issue should be fixed.