cyrils/renogy-bt #86: Trouble getting Requirements programs

Jrmason25, 2024-10-29:
After Thonny installs and I try to install plugins via requirements.txt it fails on python dbus. It's giving me a meta error. I try doing each plugin one by one first 3 install and fails on python dbus. I tried on win11 and Ubuntu vm. Same error on both. What am I doing wrong?  Thank you for any advice.

Jrmason25, 2024-10-29:
Error after trying with requirements.txt

install --user -r /home/admin/Downloads/renogy-bt-3.2/requirements.txt
Requirement already satisfied: gatt in ./.local/lib/python3.10/site-packages (from -r /home/admin/Downloads/renogy-bt-3.2/requirements.txt (line 1)) (0.2.7)
Requirement already satisfied: configparser in ./.local/lib/python3.10/site-packages (from -r /home/admin/Downloads/renogy-bt-3.2/requirements.txt (line 2)) (7.1.0)
Requirement already satisfied: requests in ./.local/lib/python3.10/site-packages (from -r /home/admin/Downloads/renogy-bt-3.2/requirements.txt (line 3)) (2.32.3)
Requirement already satisfied: paho-mqtt in ./.local/lib/python3.10/site-packages (from -r /home/admin/Downloads/renogy-bt-3.2/requirements.txt (line 4)) (2.1.0)
Collecting dbus-python
  Using cached dbus-python-1.3.2.tar.gz (605 kB)
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [20 lines of output]
      + meson setup /run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d /run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d/.mesonpy-sqkxwd9x -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d/.mesonpy-sqkxwd9x/meson-python-native-file.ini
      The Meson build system
      Version: 1.6.0
      Source dir: /run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d
      Build dir: /run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d/.mesonpy-sqkxwd9x
      Build type: native build
      Project name: dbus-python
      Project version: 1.3.2
      
      ../meson.build:4:0: ERROR: Unknown compiler(s): [['cc'], ['gcc'], ['clang'], ['nvc'], ['pgcc'], ['icc'], ['icx']]
      The following exception(s) were encountered:
      Running `cc --version` gave "[Errno 2] No such file or directory: 'cc'"
      Running `gcc --version` gave "[Errno 2] No such file or directory: 'gcc'"
      Running `clang --version` gave "[Errno 2] No such file or directory: 'clang'"
      Running `nvc --version` gave "[Errno 2] No such file or directory: 'nvc'"
      Running `pgcc --version` gave "[Errno 2] No such file or directory: 'pgcc'"
      Running `icc --version` gave "[Errno 2] No such file or directory: 'icc'"
      Running `icx --version` gave "[Errno 2] No such file or directory: 'icx'"
      
      A full log can be found at /run/user/1000/snap.thonny/pip-install-45b0bm6u/dbus-python_7c2a75c187604dffb48cb3d059c1413d/.mesonpy-sqkxwd9x/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Process returned with code 1

cyrils, 2024-10-30:
have you tried the [bleak](https://github.com/cyrils/renogy-bt/tree/bleak) branch?

Jrmason25, 2024-10-30:
Thank you for your reply. I just tried with the Bleak branch and now this error. Something about output not empty. It was a fresh esp32 micropython flash and I selected the erase all flash option during this process. Any suggestions? 

install -r C:/Users/Jeremy/Downloads/renogy-bt-bleak/renogy-bt-bleak/requirements.txt
Looking in indexes: http://127.0.0.1:36628
Requirement already satisfied: bleak in c:\users\jeremy\appdata\local\pipkin\cache\workspaces\5f5a6c731a2d621e3ba0aab6ea9f73fe\lib\site-packages (from -r C:/Users/Jeremy/Downloads/renogy-bt-bleak/renogy-bt-bleak/requirements.txt (line 1)) (0.22.3)
127.0.0.1 - - [30/Oct/2024 12:41:40] "GET /configparser/ HTTP/1.1" 200 -
Collecting configparser
  Using cached http://127.0.0.1:36628/configparser/configparser-7.1.0-py3-none-any.whl/
Requirement already satisfied: requests in c:\users\jeremy\appdata\local\pipkin\cache\workspaces\5f5a6c731a2d621e3ba0aab6ea9f73fe\lib\site-packages (from -r C:/Users/Jeremy/Downloads/renogy-bt-bleak/renogy-bt-bleak/requirements.txt (line 3)) (0.10.0)
127.0.0.1 - - [30/Oct/2024 12:41:40] "GET /paho-mqtt/ HTTP/1.1" 200 -
Collecting paho-mqtt
  Using cached http://127.0.0.1:36628/paho-mqtt/paho_mqtt-2.1.0-py3-none-any.whl/
Requirement already satisfied: typing-extensions>=4.7.0 in c:\users\jeremy\appdata\local\pipkin\cache\workspaces\5f5a6c731a2d621e3ba0aab6ea9f73fe\lib\site-packages (from bleak->-r C:/Users/Jeremy/Downloads/renogy-bt-bleak/renogy-bt-bleak/requirements.txt (line 1)) (4.12.2)
127.0.0.1 - - [30/Oct/2024 12:41:41] "GET /async-timeout/ HTTP/1.1" 200 -
Collecting async-timeout<5,>=3.0.0
  Using cached http://127.0.0.1:36628/async-timeout/async_timeout-4.0.3-py3-none-any.whl/
Requirement already satisfied: bleak-winrt<2.0.0,>=1.2.0 in c:\users\jeremy\appdata\local\pipkin\cache\workspaces\5f5a6c731a2d621e3ba0aab6ea9f73fe\lib\site-packages (from bleak->-r C:/Users/Jeremy/Downloads/renogy-bt-bleak/renogy-bt-bleak/requirements.txt (line 1)) (1.2.0)
Installing collected packages: paho-mqtt, configparser, async-timeout
Successfully installed async-timeout-4.0.3 configparser-7.1.0 paho-mqtt-2.1.0
Starting to apply changes to the target.
Copying configparserError (ManagementError(...), 'Command output was not empty')

cyrils, 2024-10-30:
Ah you are using micropython! This do not support micropython, only cpython. However checkout the ESPHome port listed in the README.

Jrmason25, 2024-11-01:
@cyrils Thank you for pointing me in the right direction. I used mavenius bt renogy to get my esp32 and home assistant working with Renogy controller. Thanks again!

closed 2024-11-02 by cyrils