cyrils/renogy-bt #96: RPI Zero 2w (32 bit Bookworm) + Renogy pro 12V BT Battery  -->  Error connecting to device

daimonionnn, 2025-01-03:
Trying to connect my RPI Zero 2W with RPI OS (Bookworm 32 bit) to Renogy Smart BT battery ( RBT12100LFP ).
I always get this error, tried reboots, upgrades, but no luck.

```
pi@zero:~/renogy/renogy-bt $ sudo python3 ./example.py config.ini
INFO:root:Init BatteryClient: RBT12100LFP-BT => 4c:e1:74:49:fc:fb
INFO:root:Starting discovery...
INFO:root:Devices found: 2
INFO:root:Found matching device RNGPRO12BATT50002965 => 4C:E1:74:49:FC:FB
INFO:root:Client connection: True
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
ERROR:root:Error connecting to device
ERROR:root:Connection failed: (<class 'bleak.exc.BleakDBusError'>, BleakDBusError('org.bluez.Error.NotSupported', 'Operation is not supported'), <traceback object at 0x73ec61c8>)
ERROR:root:on_error: (<class 'bleak.exc.BleakDBusError'>, BleakDBusError('org.bluez.Error.NotSupported', 'Operation is not supported'), <traceback object at 0x73ec61c8>)
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_write_value failed Multiple Characteristics with this UUID, refer to your desired characteristic by the `handle` attribute instead.
INFO:root:Exit: Disconnecting device: RNGPRO12BATT50002965 4C:E1:74:49:FC:FB
```

In parralel terminal I run 
`sudo busctl monitor org.bluez`

and there is this error:

Type=method_call  Endian=l  Flags=0  Version=1 Cookie=3  Timestamp="Fri 2025-01-03 13:34:54.318671 UTC"
  Sender=:1.56  Destination=org.bluez  Path=/org/bluez/hci0/dev_4C_E1_74_49_FC_FB/service002e/char003b  Interface=org.bluez.GattCharacteristic1  Member=StartNotify
  UniqueName=:1.56
  MESSAGE "" {
  };

â€Ł Type=error  Endian=l  Flags=1  Version=1 Cookie=274  ReplyCookie=3  Timestamp="Fri 2025-01-03 13:34:54.318807 UTC"
  Sender=:1.7  Destination=:1.56
  ErrorName=org.bluez.Error.NotSupported  ErrorMessage="Operation is not supported"
  UniqueName=:1.7
  MESSAGE "s" {
          STRING "Operation is not supported";
  };

Any ideas ?

cyrils, 2025-01-30:
Here is the potential fix https://github.com/hbldh/bleak/discussions/659, but I'm yet to integrate it.

I do not have the said battery, so its hard to test. It looks like the battery has duplicate characteristics with same uuid, which should not happen ideally.

cyrils, 2025-04-10:
That response is expected from Renogy. Doesn't mean it won't work with modbus. From the error it looks like the issue is: 

`Multiple Characteristics with this UUID, refer to your desired characteristic by the handle attribute instead.`

Which is unusual for a device. The fix is to also look also at the service UUID, not just rely on characteristic UUID. I just haven't  gotten around doing it.

Sparhawk76, 2025-04-13:
nowcodingaway, I saw your question to me but couldn't reply via email and it didn't show on github.    Its still working for me, I have 3 of the Renogy Pro 200Ah batteries that I monitor with it. 

Below is my BaseClient.py with my modifications, as mentioned in the other thread I use node-red to multiply the reported voltage by 10.   From the code snipped I saw that you posted you have a far more elegant solution to this.    I was using node-red (plugin version integrated into SignalK marine software) to convert the mqtt to SignalK anyways ... so was simple for me to add the conversion there.  I have yet to pull down a more recent version of renogy-bt, so haven't tested changes with the current code base.   

As for my firmware versions on my batteries, I'll have to disable my monitoring scripts for a bit and check with the renogy app ... I don't really use it.  If you need it I can look, just don't have the time right now as I type this (10:25pm local time).

`import os
from threading import Timer
import logging
import configparser
import time
from .Utils import bytes_to_int, int_to_bytes, crc16_modbus
from .BLE import DeviceManager, Device

# Base class that works with all Renogy family devices
# Should be extended by each client with its own parsers and section definitions
# Section example: {'register': 5000, 'words': 8, 'parser': self.parser_func}

ALIAS_PREFIX = 'BT-TH'
ALIAS_PREFIX2 = 'RNGRBP'
NOTIFY_CHAR_UUID = "0000fff1-0000-1000-8000-00805f9b34fb"
WRITE_CHAR_UUID  = "0000ffd1-0000-1000-8000-00805f9b34fb"
READ_TIMEOUT = 30 # (seconds)

class BaseClient:
    def __init__(self, config):
        self.config: configparser.ConfigParser = config
        self.manager = None
        self.device = None
        self.poll_timer = None
        self.read_timer = None
        self.data = {}
        self.device_id = self.config['device'].getint('device_id')
        self.sections = []
        self.section_index = 0
        logging.info(f"Init {self.__class__.__name__}: {self.config['device']['alias']} => {self.config['device']['mac_addr']}")

    def connect(self):
        self.manager = DeviceManager(adapter_name=self.config['device']['adapter'], mac_address=self.config['device']['mac_addr'], alias=self.config['device']['alias'])
        self.manager.discover()

        if not self.manager.device_found:
            logging.error(f"Device not found: {self.config['device']['alias']} => {self.config['device']['mac_addr']}, please check the details provided.")
            for dev in self.manager.devices():
#                if dev.alias() != None and dev.alias().startswith(ALIAS_PREFIX):
                if dev.alias() != None and ( dev.alias().startswith(ALIAS_PREFIX) or dev.alias().startswith(ALIAS_PREFIX2) ):
                    logging.debug(f"Possible device found! ======> {dev.alias()} > [{dev.mac_address}]")
            self.__stop_service()

        self.device = Device(mac_address=self.config['device']['mac_addr'], manager=self.manager, on_resolved=self.__on_resolved, on_data=self.on_data_received, on_connect_fail=self.__on_connect_fail, notify_uuid=NOTIFY_CHAR_UUID, write_uuid=WRITE_CHAR_UUID)

        try:
            self.device.connect()
            self.manager.run()
        except Exception as e:
            self.__on_error(True, e)
        except KeyboardInterrupt:
            self.__on_error(False, "KeyboardInterrupt")

    def disconnect(self):
        self.device.disconnect()
        self.__stop_service()

    def __on_resolved(self):
        logging.info("resolved services")
        self.poll_data() if self.config['data'].getboolean('enable_polling') == True else self.read_section()

    def on_data_received(self, response):
        self.read_timer.cancel()
        operation = bytes_to_int(response, 1, 1)
[... 53 lines cut ...]
            data.append(int_to_bytes(regAddr, 0))
            data.append(int_to_bytes(regAddr, 1))
            data.append(int_to_bytes(readWrd, 0))
            data.append(int_to_bytes(readWrd, 1))

            crc = crc16_modbus(bytes(data))
            data.append(crc[0])
            data.append(crc[1])
            logging.debug("{} {} => {}".format("create_request_payload", regAddr, data))
        return data

    def __on_error(self, connectFailed = False, error = None):
        logging.error(f"Exception occured: {error}")
        self.__stop_service() if connectFailed else self.disconnect()

    def __on_connect_fail(self, error):
        logging.error(f"Connection failed: {error}")
        self.__stop_service()

    def __stop_service(self):
        if self.poll_timer is not None and self.poll_timer.is_alive():
            self.poll_timer.cancel()
        if self.poll_timer is not None: self.read_timer.cancel()
        self.manager.stop()
        os._exit(os.EX_OK)`

Sparhawk76, 2025-04-13:
Anoying ... it broke up the code I tried to post.  Don't have time to figure it out right now, too late at night.   Will try and find time tomorrow to post a cleaner version.

cyrils, 2025-04-13:
@nowcodingaway Instead of this `write_char = write_candidates[-1]` the better way is to identify the characteristic by the correct service. Do you know which service is hosting the correct char?

cyrils, 2025-04-13:
Can you try to use this branch and see if it works? [char-handle](https://github.com/cyrils/renogy-bt/tree/char-handle)

cyrils, 2025-04-20:
Fixed with https://github.com/cyrils/renogy-bt/pull/106

closed 2025-04-20 by cyrils