cyrils/renogy-bt #97: Can't poll/pair for BT device if polling mode is force closed

MrBryan, 2025-01-03:
If I ever close the process that is happily polling my bt-2, it won't work when i start it up again.  

Only way to fix is go to the BT-2 and plug and replug it.

I also can not use the native renogy app at this stage. It seems it ties up the BT pairing or something like this.  Perhaps there is a way to flush it? 

(I hid the mac address belwo)

`INFO:root:BT-HIDDEN     => {'function': 'READ', 'model': 'RNG-CTRL-RVR30', 'device_id': 16, 'battery_percentage': 42, 'battery_voltage': 12.0, 'battery_current': 0.4, 'battery_temperature
ontroller_temperature': 8, 'load_status': 'off', 'load_voltage': 0.0, 'load_current': 0.0, 'load_power': 0, 'pv_voltage': 26.4, 'pv_current': 0.18, 'pv_power': 5, 'max_charging_power_today': 7
scharging_power_today': 0, 'charging_amp_hours_today': 1, 'discharging_amp_hours_today': 0, 'power_generation_today': 12, 'power_consumption_today': 0, 'power_generation_total': 953, 'charging
 'mppt', 'battery_type': 'sealed', '__device': 'rover-li', '__client': 'RoverClient'}
pvoutput <Response [200]>
^CERROR:root:Exception occured: KeyboardInterrupt
ERROR:root:on_error: KeyboardInterrupt
^C
mrbryan@raspberrypi:~/renogy-bt $ python3 ./example.py config.ini
INFO:root:Init RoverClient: rover-li => HIDDENMAC
INFO:root:Starting discovery...
INFO:root:Devices found: 6
ERROR:root:Device not found: rover-li => HIDDENMAC, please check the details provided.`

MrBryan, 2025-01-03:
seems if i do 
```
sudo hciconfig hci0 reset
sudo invoke-rc.d bluetooth restart
```
on my raspberry pi zero 2 w

that lets it find it again.. but maybe some kind of routine to unpair should be built in when a term signal is received

Cheers

cyrils, 2025-01-03:
Looks like it won't disconnect clean on KeyboardInterrupt, but it should.

milperks, 2025-04-06:
> seems if i do
> 
> ```
> sudo hciconfig hci0 reset
> sudo invoke-rc.d bluetooth restart
> ```
> 
> on my raspberry pi zero 2 w
> 
> that lets it find it again.. but maybe some kind of routine to unpair should be built in when a term signal is received
> 
> Cheers

Thanks I was having the same problem. I might just set this up to run every couple of days on a cron job.

milperks, 2025-04-06:
> My challenge is this then breaks other device connections. And if they don't have recovery - they now have a problem.
> 
> And if baking it into the script - if you have 2 instances or more for different devices they start to fight each other by resetting the adapter.
> 
> I've been trying to dig into this as recently as yesterday. There seems to be a problem with how bleak (or rather bluepy) is handing the connection around. I've yet to find a reliable way to clear out an old connection without effectively turning the adapter off and on again.
> 
> I have been down rabbit holes of how connection caching works in bleak and all sorts.
> 
> But even the most hacky things don't seem to do it. Either via interrupt or even running a whole bunch of connection clean up at initialisation before scanning.
> 
> If I have any luck any time. I'll share. But I've been trying to sort this for months and as a non coder it's just me, search and trying things as best I can.

Ah, I am using a pi zero 2 solely for this script (I couldn't figure out how to get it to run natively in home assistant on my Raspberry pi), so I can afford to just restart the Bluetooth connection.

I'm also a non-coder, have you tried unpairing and re-pairing it in bluetoothctrl? I unpaired and then rebooted but that didn't solve it.

milperks, 2025-04-06:
> > > My challenge is this then breaks other device connections. And if they don't have recovery - they now have a problem.
> > > And if baking it into the script - if you have 2 instances or more for different devices they start to fight each other by resetting the adapter.
> > > I've been trying to dig into this as recently as yesterday. There seems to be a problem with how bleak (or rather bluepy) is handing the connection around. I've yet to find a reliable way to clear out an old connection without effectively turning the adapter off and on again.
> > > I have been down rabbit holes of how connection caching works in bleak and all sorts.
> > > But even the most hacky things don't seem to do it. Either via interrupt or even running a whole bunch of connection clean up at initialisation before scanning.
> > > If I have any luck any time. I'll share. But I've been trying to sort this for months and as a non coder it's just me, search and trying things as best I can.
> > 
> > 
> > Ah, I am using a pi zero 2 solely for this script (I couldn't figure out how to get it to run natively in home assistant on my Raspberry pi), so I can afford to just restart the Bluetooth connection.
> > I'm also a non-coder, have you tried unpairing and re-pairing it in bluetoothctrl? I unpaired and then rebooted but that didn't solve it.
> 
> Technically the device is never paired as it's a BLE device type. So it doesn't have the same flags as far as I can tell / understand.
> 
> I've made my connection processes much much better. And if I turn the adapter itself off and on - even via the hitools etc I have an almost perfect connection success rate.
> 
> I may have to also end up getting 3 pi zero (one for each renogy device) but I don't feel fully defeated just yet :)

I wouldn't have thought it was paired either but when I run Bluetoothctl and then devices, it's listed.

Also strangley, when it was in this cached connection state, Bluetoothctl was showing as that device name. Once it had successfully disconnected it went back to just 'bluetooth'. 

[image]
Trying to disable scanning gave me an error so ignore the extra info. 

When this reoccurs I'll have another play around with it, but it seems like you are much further along than me!

milperks, 2025-04-06:
> I'm basically just hitting things with hammers until it works :)
> 
> But I'm feeling ready to have another go at this today with Claude.
> 
> I've created a fully comprehensive clean up script that also cleans up at the DBus level.
> 
> I'm going to mod BLE manager script to capture details about successful connection properties and log those.
> 
> Then I'm going to have my cleanup script run at init along with killing based on the meta data I gather regardless of if I discover the device is connected or not.
> 
> I've got logging at each stage so I might be able to discover which cleanup is working (if any do) and can then optimise and test that approach.
> 
> If I have success I'll share what I find.

Hey, I've found if you run bluetoothctl then remove the device it will work again, 'remove [mac address]'. 

I'm away from home and just on my phone so can't figure out at the moment how that can be added into the script but if it can be then it should work!

cyrils, 2025-04-10:
I have a branch [clean-exit](https://github.com/cyrils/renogy-bt/tree/clean-exit) with a fix. Can you guys try it out and let me know?

cyrils, 2025-04-20:
Fixed with https://github.com/cyrils/renogy-bt/pull/107

closed 2025-04-20 by cyrils