cyrils/renogy-bt #99: what am i missing

theboss0539, 2025-01-19:
admin@raspberrypi:~/renogy-bt $ python3 ./example.py config.ini
INFO:root:Init BatteryClient: BT-TH-161D2826C => AC:4D:16:1D:28:6C
INFO:root:Starting discovery...
INFO:root:Devices found: 8
INFO:root:Found matching device BT-TH-161D286C     => AC:4D:16:1D:28:6C
INFO:root:Client connection: True
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_write_value succeeded
INFO:root:notification_callback
WARNING:root:on_data_received: unknown operation=0

cyrils, 2025-01-20:
Whats your battery model?

theboss0539, 2025-01-20:
4 RBT50LFP48S daisy chained to one bt-2

cyrils, 2025-01-20:
if daisy chained you cannot use the device id 255. Follow the instructions in the readme to find the individual device ids.

theboss0539, 2025-01-20:
I have tried connecting on battery at a time and I get
admin@raspberrypi:~/renogy-bt $ python3 ./example.py config.ini
INFO:root:Init BatteryClient: BT-TH-161D2826C => AC:4D:16:1D:28:6C
INFO:root:Starting discovery...
INFO:root:Devices found: 10
INFO:root:Found matching device BT-TH-161D286C     => AC:4D:16:1D:28:6C
INFO:root:Client connection: True
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_write_value succeeded
ERROR:root:on_read_timeout => Timed out! Please check your device_id!
INFO:root:Exit: Disconnecting device: BT-TH-161D286C     AC:4D:16:1D:28:6C

And I have tried all devices id's listed online , also I have had it working with this set up before

theboss0539, 2025-01-20:
my problem was a keyboard interrupt fixed by rebooting pi

theboss0539, 2025-01-20:
ok so home assistant can see the mqtt packet in mqtt configuration when listening to batt/state but i cant not seem to get the battery soc or solar power to update 

my config.yaml
mqtt:
  sensor:
    - name: "Solar Power"
      state_topic: "batt/state"
      device_class: "voltage"
      unit_of_measurement: "V"
      value_template: "{{ value_json.pv_power }}"
    - name: "Battery SOC"
      state_topic: "batt/state"
      device_class: "battery"
      unit_of_measurement: "%"
      value_template: "{{ value_json.battery_percentage }}"
# check output log for more fields  

also is there a way to change the battery id as i have 4 batteys with ids 48 , 247, 48, 247 and can only get data from 2 
this a a side conncern output to ha is main