cyrils/renogy-bt #108: Connection to battery RBT12200LFP-BT fails.

nikirikitiki, 2025-04-24:
Hi! I am trying to poll **RBT12200LFP-BT**. However, for some reason it fails after establishing the connection:

`INFO:root:Init BatteryClient: RNGRBP1217000105 => 7231D56C-B0AD-E65A-BA1F-D1DAF716240D
INFO:root:Starting discovery...
INFO:root:Devices found: 8
INFO:root:Found matching device RNGRBP1217000105 => 7231D56C-B0AD-E65A-BA1F-D1DAF716240D
INFO:root:Client connection: True
INFO:root:found write characteristic 0000ffd1-0000-1000-8000-00805f9b34fb, service 0000ffd0-0000-1000-8000-00805f9b34fb
INFO:root:subscribed to notification 0000fff1-0000-1000-8000-00805f9b34fb
ERROR:root:Error connecting to device
ERROR:root:Connection failed: (<class 'bleak.exc.BleakError'>, BleakError('Failed to update the notification status for characteristic 59: Error Domain=CBATTErrorDomain Code=6 "The request is not supported." UserInfo={NSLocalizedDescription=The request is not supported.}'), <traceback object at 0x106ad5100>)
ERROR:root:on_error: (<class 'bleak.exc.BleakError'>, BleakError('Failed to update the notification status for characteristic 59: Error Domain=CBATTErrorDomain Code=6 "The request is not supported." UserInfo={NSLocalizedDescription=The request is not supported.}'), <traceback object at 0x106ad5100>)
INFO:root:writing to 0000ffd1-0000-1000-8000-00805f9b34fb [255, 3, 19, 136, 0, 17, 20, 182]
INFO:root:characteristic_write_value succeeded
INFO:root:Exit: Disconnecting device: RNGRBP1217000105 7231D56C-B0AD-E65A-BA1F-D1DAF716240D
`

There is no issue when connecting to DC-DC charger with BT-2 Bluetooth Module.

Do you have advice on which way to dig? 
Could it be wrong UUID/handle for the device?

nikirikitiki, 2025-05-03:
After some investigation - I found that the issue was with the 
`NOTIFY_CHAR_UUID = "0000fff1-0000-1000-8000-00805f9b34fb"` inside `BaseClient.py`

The battery uses different UUIDs, so instead you need to expand it to a list in `BaseClient.py`:
`NOTIFY_CHAR_UUID = ["0000fff1-0000-1000-8000-00805f9b34fb", "0000fff1-0000-1000-8000-00805f9b34fb", "00006487-3c17-d293-8e48-14fe2e4da212"]`

Then adjust the `BLEManager.py` to accommodate a starting at lines **42**:

```
for service in self.client.services:
    for characteristic in service.characteristics:
        if characteristic.uuid in self.notify_char_uuid:
            try:
                await self.client.start_notify(characteristic,  self.notification_callback)
                logging.info(f"subscribed to notification {characteristic.uuid}")
            except Exception as e:
                logging.error(f"Error subscribing to notification: {e}")

        if characteristic.uuid == self.write_char_uuid and service.uuid == self.write_service_uuid:
            self.write_char_handle = characteristic.handle
            logging.info(f"found write characteristic {characteristic.uuid}, service {service.uuid}")
```

cyrils, 2025-05-04:
Mid creating a pull request to branch [battery-pro](https://github.com/cyrils/renogy-bt/tree/battery-pro)?